kennetek/gridfinity-rebuilt-openscad #13: 50mm grid issues

peter-banks, 2022-09-27:
Thanks so much for this project, its exactly what ive been looking for, however ive got a couple minor issues with the dimensions when I'm running at 50mm grid x/y/z size. I'm using the original frames scaled to 50mm increments and the original divider boxes scaled to the same work fine.

1. The radius of the base where it sits in the grid is slightly too "square", once I filed it down a bit it fits well. I tried playing with the parameters but couldn't quite find the right one.
2. The lip with the tab is about 1mm too high and means they dont stack as well together.
3. If i set `length = 50` the base is slightly too big as is the lip opening so they dont stack as nicely, if I set to 49 everything fits well. This workaround is fine for my needs but just mentioning it anyway.

Attached the customerize config
[image]

unteins, 2022-09-27:
Assuming that the 42 mm bins ARE perfectly the same as the original, it would probably make the most sense to simply make a bin at 42 mm grid size and scale it up to 50 mm as you do with other bins. This should solve all of the small quirks from trying to make them natively at 50 mm.

peter-banks, 2022-09-28:
Thank you @kennetek and @unteins. 

@kennetek Yeah sorry i meant Zack's bins and frames from Thangs.
Both solutions technically work and perfectly usable for my purpose however changing the length (as you mentioned) does seem to affect the hole placement and size (see the photo below, bin marked K). I think using @unteins idea keeps it scaled properly and less likely to hit other issues so probably the safest option in this case.

Attaching my openscad config for reference
[image]
[image]

closed 2022-09-28 by peter-banks