kennetek/gridfinity-rebuilt-openscad #22: Configurable tabs and notches in base plates for easier alignment

gudvinr, 2022-11-06:
Often you have a need to print base smaller than you need, e.g. for drawer which size is bigger than your printer bed.  
When you print these parts, it is tricky to glue everything together aligned.

Simple bumps on edge of one part and corresponding notches on another part make joining process much simpler and reliable.

Having these joints generated helps with consistency compared to process of adding them manually.

khardix, 2022-12-25:
To expand on this idea, another option I would like to see is something similar to [screw-together baseplate](https://www.printables.com/cs/model/300603-gridfinity-screw-together-baseplate) – holes around the walls of the baseplate for (M3) screw heads and sockets/heat-set inserts. That will enable both aligning the parts correctly and attaching them without using glue.

Granted, I'm speaking about the skeletonized baseplate here – not sure if this is applicable to the other variants, while the notches probably are.

jazzhands, 2022-12-27:
See my just created [pull request](https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/39) for screw-together baseplate functionality.  

As you note, it does only work on skeletonized baseplates.

This doesn't completely answer @gudvinr 's request, but it solves it for skeletonized baseplates. You can use screws or dowels for alignment.