kennetek/gridfinity-rebuilt-openscad #23: Vase mode trays

gudvinr, 2022-11-06:
Default bins are very durable but they require a ton of plastic and it takes quite a lot of time to print them.

On the other hand, with bigger nozzle vase mode allows you to print relatively sturdy boxes. As a bonus, you don't have seam line which gives nicer finish.

Plain box without curved bottom and label notch is relatively simple to make in vase mode.  
Curved bottom is impossible to print without making compromises to stability.  
Label notch is tricky to print in default position but not impossible if you remove back side of the box and make top side angled.

kennetek, 2022-11-06:
I am a bit confused about what you are talking about, are you suggesting a new feature, or do you have a problem with the existing [vase mode implementation](https://github.com/kennetek/gridfinity-rebuilt-openscad/tree/main/gridfinity-spiral-vasel)?

gudvinr, 2022-11-06:
I feel stupid now because I didn't see separate folder for some reason and it isn't mentioned in README.

However, after poking around I see that vase mode bins printed in 2 parts with separate base. I kinda understand why it's done that way but I also think that it is possible to print simple bins (that do not require magnets, for example) to be printed without making base separate piece.

kennetek, 2022-11-06:
Vase mode is in a separate folder because I wanted a separate readme, due to it being more complicated and has specific instructions. I also realize that it was not mentioned in the main readme, so I fixed that. I think the documentation needs an upgrade though, as the number of features is becoming overwhelming for a single document. 

Its not the magnets that require a separate base piece, its the bases themselves. For instance, a 2x2 is not really possible, given how there needs to be indents in the base to fit on the grid. Therefore, only 1x1s are possible, and I think there already exist multiple designs for 1x1 grids. For [example](https://thangs.com/designer/LittleHobbyShop/3d-model/%23gridfinity%20Vase%20Mode%20Single%20Box-65828). Given how that is only a single file, I didn't see a need to redo it in OpenSCAD.

gudvinr, 2022-11-06:
Okay, that sounds reasonable.

I thought about generating gcode for base and rest of the box separately then join them in slicer but I found 2 issues. Probably worth filing another issue for each though because they don't seem to be related:
1. If you set "style base" to "none", cross indeed disappear from the _bin_ bottom but there is still elevated cross on the bottom part which I think makes these parts incompatible
2. If you set "bottom layer" to 0 (to avoid printing bottom since it'll be joined anyway), chamfer disappear from the base.

kennetek, 2022-12-11:
1. I guess the cross should also be removed from the base block, it can't hurt. But I was more envisioning if someone just wanted a vase mode bin for some reason, so they wanted a flat bottom. I think it would be a bit hard to get the proper glue-up without the alignment crosses, but I guess it is possible. I'll fix that soon. 
2. Is there a use case for removing the entire bottom layer? That was definitely something I did not imaging, since it does not seem practical.

gudvinr, 2022-12-11:
> Is there a use case for removing the entire bottom layer? That was definitely something I did not imagine, since it does not seem practical.

If you use stls to produce gcode in slicer (separately for top and bottom) and then join top gcode on top of base gcode manually, this bottom layer serves no purpose since you don't need to have any sort of alignment.

> I think it would be a bit hard to get the proper glue-up without the alignment crosses

You are correct, but both points come from same use case where you join gcodes so you don't need to glue parts together later.

commit 6804bf99 (2023-02-06): Fixed #23

Chamfer section was not touching the bottom layers. Maybe this was causing problems with slicing in Cura
closed 2023-02-06 by kennetek