kennetek/gridfinity-rebuilt-openscad #24: Scoop chamfer error

rox97, 2022-11-21:
[image]

Trying to print the bin with Scoop chamfer activated and 2 compartments, one of the two has an error.
Printing the one with one compartment, there is no error

[image]

kennetek, 2022-11-21:
[image]

I investigated, and I'm relieved to say that it looks like a slicer issue, not a model issue. The vase mode slicing for these bins is somewhat hacky, so it might need a bit of fiddling to get it to work. I did the same thing you did, and replicated the issue, but it was only one layer of error instead of multiple. I think it is because the diagonal seam, shown here, is intersecting that side, and it doesn't like that. I was able to get it to go away by increasing/decreasing the layer height in the slicer by 0.01 mm (0.29 or 0.31 if the OpenSCAD script has 0.3). I think that accidentally relocates the seam to a less critical part of the bin. As far as I know, there is not a way to specify what part of the bin has that diagonal seam.

rox97, 2022-11-21:
Thank you. I see that actually the problem is probably the seam. I tried changing the layer height in the slicer but couldn't move the seam. I also tried changing the seam positioning option of prusaslicer or the seam painting function, but to no avail. Did you use prusaslicer or cura?

rox97, 2022-11-21:
I also tried creating the stl with a layer height of 0.3 and in the slicer I get an error where it detects empty layers, even changing the layer height of +-0.01 in the slicer. 
Probably these two problems are not related to each other.
[image]

rox97, 2022-11-21:
I don't know how but i managed to change the seam position, but the problem is always there, even changing the slicer layer height
[image]

kennetek, 2022-11-21:
[image]

Even though you got the seam on the back, I think the intersection with the middle divider is still messing with the slicing. Try to get the seam on the side walls. 

Other than that advice, I don't know if I can help much. I'm very interested in the solution, however.

kennetek, 2023-02-06:
I recently looked at vase mode again, and realized the scoop chamfer was not touching the bottom of the bin. This may have been causing the issue? In one of my screenshots, the bottom of the chamfer is blue, as its bridging in air. I fixed that issue, but it needs to be tested to see if it slices better.

rox97, 2023-02-06:
thanks for working on it! it seems that the modification solves some problems in the slicing, but there is still an error. it is probably caused by the position of the seam. I've also tried changing the position of the seam, both in the settings and using the seam paint function, and also changing the position of the object on the plane, but the result doesn't change. Later I try to print it to see what comes out
[image]
[image]
[image]

rox97, 2023-02-06:
Forgive me, but I deleted the comment I had just written. I just noticed that I had set the height of the layers wrong. setting the right one, the slicing seems to give no problems. Thank you for working on it! I'll try to print later

closed 2023-02-14 by kennetek