kennetek/gridfinity-rebuilt-openscad #36: lite bins base is only 0.4ish mm thick

patrickmcquay, 2022-12-16:
I am not sure how to solve this, but it looks like there is an issue with the offset code in block_base_solid, which doesnt offset the bottom of the base upwards enough to satisfy the given offset.

This results in the default lite bin having a very thin (since I'm printing these with 0.3mm layers, one layer) bottom.

Is this intentional? Can I adjust it with the settings that are already here?

dmgerman, 2023-01-26:
i ran into the same problem. Since I didn't want to waste the boxes, I used a bit of crazy glue to reinforce the bottom seams. I looked at the code but it is not obvious where that offset comes from. @kennetek if you point us in the right direction, we might be able to help fixing/testing the code. Thank you nonetheless for making it available.

dmgerman, 2023-01-27:
it seems that the problem is the last rounded_rectangle call in gridfinityLite. It flushes the top of the ridges, but just exactly at 5mm+0.5 (tolerance). if the translation just above gets added 1mm, it seems to solve the issue.

Larswa, 2023-02-14:
@dmgerman   Where exactly did you add the 1mm?  Not an openscad dev myself, but I can kinda see whats going on. Have tried adding 1mm to almost everything in the last translate before the last rounded_rectangle call, but couldnt get it to make a difference. 

> it seems that the problem is the last rounded_rectangle call in gridfinityLite. It flushes the top of the ridges, but exactly at 5mm+0.5 (tolerance). if the translation just above gets added 1mm, it seems to solve the issue.

```
difference() {
            union() {
                intersection() {
                    difference() {
                        gridfinityBase(gridx, gridy, length, div_base_x, div_base_y, style_hole, -d_wall*2, false);
                        translate([-gridx*length/2,-gridy*length/2,2*h_base])
                        cube([gridx*length,gridy*length,1000]);
                    }
                    translate([0,0,-1])
                    rounded_rectangle(gridx*length-0.5005-d_wall*2, gridy*length-0.5005-d_wall*2, 1000, r_f2);
                }
                translate([0,0,h_base+d_clear])
                rounded_rectangle(gridx*length-0.5005-d_wall*2, gridy*length-0.5005-d_wall*2, h_base, r_f2);
            }

            translate([0,0,-4*h_base])
            gridfinityInit(gridx, gridy, height(20,0), 0, length)
            children();
        }
```

commit ba3c3e15 (2023-02-15): Fixed #36

Added parameter to control bottom layer thickness
closed 2023-02-15 by kennetek