kennetek/gridfinity-rebuilt-openscad #47: Idea: add option to remove the inner jetties of skeletonized baseplates along the longer side

LorenzKahl, 2023-01-09:
If checked the marked areas would be dropped.
[image]

This could save some filament without impacting stability a lot - I guess.
Unfortunately I'm a total OpenSCAD NOOB so unfortunately I can't chip in a PR for this. I also assume that the required logic for this is not quiet that easy.

Ruudjhuu, 2023-01-17:
Interesting idea, however, I do think it will impact stability a lot if the baseplate gets bigger. I created a POC which only works in the x direction for anyone to try.

[poc/inner_jeties branch in Ruudjhuu/gridfinity-rebuilt-openscad](/Ruudjhuu/gridfinity-rebuilt-openscad/tree/poc/inner_jeties)

LorenzKahl, 2023-01-18:
Hi @Ruudjhuu, thank you, but the link seems to be dead. Could you provide another source and I'll be trying it out.

Ruudjhuu, 2023-01-18:
The link is working for me, Maybe this one works?

[https://github.com/Ruudjhuu/gridfinity-rebuilt-openscad/tree/poc/inner_jeties](https://github.com/Ruudjhuu/gridfinity-rebuilt-openscad/tree/poc/inner_jeties)

LorenzKahl, 2023-01-18:
Yeah, you were right. I was on my phone earlier and when I clicked the link it was directing me to `localhost`.
The model looks fine to me. I'm starting a print right now and will report tomorrow how it came out. The savings of filament seem to be a little less than 10% which is quiet nice ;-)

Unfortunately I can't go any bigger than 4x4 on my Mini.
[image]
[image]

LorenzKahl, 2023-01-19:
Hi @Ruudjhuu, the PoC came out really nice! I added some pictures and videos to get an impression. 
Off course, the stiffness of the whole thing reduces the bigger your prints get. But for tucking the baseplate in a drawer I don't see any problem. Also the asymmetric look kinda bothers my eye, but again - you won't even see it when it's stuffed in a drawer and stacked with bins.

I would really like to see this added to the [gridfinity-rebuilt-openscad](https://github.com/kennetek/gridfinity-rebuilt-openscad) repo 😍 

Thanks for giving this a shot!
[image]

Videos on YT:
https://youtu.be/gqlihwKOFuw
https://youtu.be/reW4JjOzt7s

kennetek, 2023-02-06:
@Ruudjhuu I'd recommend adding this as another style to go with your other minimalist baseplates.