kennetek/gridfinity-rebuilt-openscad #53: cant generate a thin baseplate

coolian1337, 2023-01-17:
[image]

it just generates it like this also when i had the baseplate larger (10 x 10 plates) it would generate the thin baseplate and this baseplate but this baseplate also only 4x4 plates

it doesnt matter wether or not i use fast-csg or not
i am on the openscad version thats currently the newest nightly (2023.0.1.17)
since i just today (2023.01.17) downloaded this i am also using the newest gridfinity-rebuilt-openscad version

Ruudjhuu, 2023-01-17:
You are drawing two models on top of eachother. By using `include`, the code outside the module definitions in `gridinity-rebuilt-baseplate.scad` is also excecuted. Try to use `use` instead of `include`

```scad
use<gridfinity-rebuilt-baseplate.scad>

gridfinityBaseplate(4, 4, 42, 0, 0, 0, true, 0);
```

coolian1337, 2023-01-18:
thank you so much!

closed 2023-01-18 by coolian1337