kennetek/gridfinity-rebuilt-openscad #56: Idea: Magnet holes without gluing.

Ruudjhuu, 2023-01-24:
Lets add a small layer on top of the magnets hole so we can pause the print, insert magnets and resume the print, This eliminates the messy gluing of magnets which might not stay in place. 

got the idea from the thread from [u/3dbender](https://www.reddit.com/r/gridfinity/comments/10kgkqw/why_gridfinity/)

oernii, 2023-03-27:
Here is something for fusion360
https://github.com/software-2/FusionGridfinityGeneratorPressureFit

Ruudjhuu, 2023-03-27:
@nelsongraca with press fit, do you mean a tighter hole? That is currently fairly easy to do by changing r_hole2 in the constants file (you can also overwrite this in any file where the utility modules are used). This also depends on how precise someones 3d printer can print. I do not think it is possible to create a boolean in the UI which just enables press fit.

mrcpu99, 2023-03-28:
I created pressfit (or more accurately print-inplace, albeit it started
with pressfit), by slightly modifying the cadquery version to leave a tiny
hole in the baseplate on the top side, that opened into the larger cavity
with a 2 layer covering, but the small hole (.5mm I think), keeps the big
cavity open.  Pause at layer X, drop in magnets, and voila.

I tried to recreate the same in the openscad one, but can't quite figure it
out, I just don't know it well enough.

I did not have good luck with pressfit and PETG plates.  Either too tight
or too loose, never  quite got it dialled in.  Might try some PLA and see,
but either I was hammering too hard and the PETG shattered, or after a few
insertions and removals of the bins, the magnet would pop out.

But I'd love to see any solution that doesn't involve gluing, especially to
PETG. :)

marc-gist, 2023-03-30:
customizable hole size would allow people to choose any of the above I would think.!

kennetek, 2023-03-30:
Like Ruud said, adjusting `r_hole2` in `gridfinity-rebuilt-constants.scad` will allow you to tune the hole size for your magnets and your printer. The main file parameters are typically reserved for high level abstract features, while the constants file has specific dimensions. 

There are a couple ways people have been looking at inserting magnets for baseplates, including pausing the print as well as adding a single layer "bar" that bridges over the gap to allow magnets to slip under.

GreenEllipsis, 2023-03-30:
I wonder if a snap-fit might be the best of both worlds?

nelsongraca, 2023-04-01:
@GreenEllipsis Sounds like a good idea

GreenEllipsis, 2023-04-04:
I am thinking now that slots might be easier to implement and more robust.
__________________________
Brian Alano
***@***.***
(812) 320-1508
*Green Ellipsis*
Design for the Circular Economy

john-parton, 2023-12-17:
I believe there is an implementation for press-fit magnets in the base here: https://www.printables.com/model/413761-gridfinity-refined