kennetek/gridfinity-rebuilt-openscad #59: easier way to get filled boxes

avxmw, 2023-01-30:
I am trying to make boxes which have the same height as for example a 1x1x6(+lip) box, but are completely filled up including the height of the lip. The intention being to have a flat surface without and boundaries to cut toolshapes from while still being same height as my stack-able containers.

[Jamie](https://github.com/vector76/gridfinity_openscad) has a 1-click option to fill the box, however it doesn't fill up to the lip (for which I filed a request).

Your version is not straight forward to do and toying around this is the best I could come up with:
[image]

This obviously has some faults, walls going down to the bottom, that slim extra ridge for some reason. Somehow the "internal block" seems to be a full gridfinity box and not a cube, too.

Maybe I misunderstand how this should be used, if so I'd be happy for some explanation. If not, all I need is a "fill" checkbox and if a lip is used the option to fill complete height including lip.

Thanks.

kennetek, 2023-02-06:
Make `divx` or `divy` zero to get a solid bin. You can also comment out the `cutEqual` module inside of the `gridfinityInit` block. This will remove the compartment cutout. 

If you want examples of how to do custom cutouts, they are at the bottom of the file. 

If you want a bin that is filled in all the way up to the tippy-top, but the same height as other 6u bins, you need to disable the lip, switch `gridz_define` to external height, and specify `7*u+3.6` as your `gridz`, where "u" is the unit height of the bin (6) and 3.6 is the height of the lip. 

I understand this is not easy to do, I never thought about people wanting to fill all the way up to the top. If this is a preferred method of making bins, then a fill toggle may be in order.

avxmw, 2023-02-06:
Thank you both for your inputs. 

What @kennetek suggests works 99%, however in the current setup openscad's customizer only takes integers for the _gridz_ field. 

I know it is an edge case, but in my opinion makes sense on the top layer of your toolbox as it provides a cleaner look,  less chance hitting the lid and most importantly for me, easier cleaning.

I'd be thankful for a version where this is easier to do, however the given instructions work for me and if you're not interested in implementing this that's obviously ok.

Thanks for the help!

kennetek, 2023-02-07:
It's not a toggle button, but by setting `style_lip` to 2 and switching either `divx` or `divy` to zero, you get a solid bin that is the same height as other bins.

closed 2023-02-07 by kennetek

avxmw, 2023-02-08:
Easy enough, thanks for this :)

commit e2e9ad53 (2023-02-07): Fixed #59

Changed `enable_lip` to `style_lip` to allow a choice between a regular lip, removal of the lip (losing 3.6mm of height) or remove lip while retaining the regular height