kennetek/gridfinity-rebuilt-openscad #64: support half-sized grid bottoms

avxmw, 2023-02-06:
[Jamie](https://github.com/vector76/gridfinity_openscad) has a feature in his version he calls _half pitch_ which is slotting the bottom into 4 equal parts to make boxes able to sit at ½ of a grid and saving some material/printing time on top.

I'd like to see something similar here.

kennetek, 2023-02-07:
I actually did this before Jamie did. You can do it by changing the `div_base_x` and `div_base_y` values. Half pitch would be 2. Any pitch is possible. Setting `gridx` or `gridy` to non integer values (like 2.5, 3&1/3) will automatically divide the bases if `div_base_x` or `div_base_y` are zero.

avxmw, 2023-02-08:
Ah damn, embarrassing, totally missed that. Thanks for pointing it out.

closed 2023-02-08 by avxmw