kennetek/gridfinity-rebuilt-openscad #67: add baseplate variant using dovetails

avxmw, 2023-02-13:
Some guys have been working on using classic dovetails to connect baseplates (https://www.printables.com/search/all?q=gridfinity+dovetail) which I think is a better solution that using screws and would thus be a nice option to have.

CumpsD, 2023-02-22:
This would be great, it would allow to build a rigid baseplate over a big surface (bigger than my 3d printer's print area)

tobycheese, 2023-09-19:
Another idea for connecting baselates together would be [snap connectors](https://www.printables.com/model/430144-gridfinity-base-with-snap-connectors).

e-noyau, 2023-12-21:
Instead of doing dovetails, an alternative approach could be to use bowties to connect plates together instead of screws. See [gridfinity refined](https://www.printables.com/model/413761-gridfinity-refined ) for a good example.

tsarna, 2023-12-21:
Bowties compatible with Gridfinity Refined would be fantastic.

cgardner-rootstock, 2024-07-31:
I'd love to see this capability for bins too.  I found [this project](https://makerworld.com/en/models/550911#profileId-469384) that works, but it's limited to 1 square at a time and a 3U height.

tobymurray, 2024-09-23:
Trivial, looks like Printables URL has changed: https://www.printables.com/search/models?ctx=models&q=gridfinity+dovetail