kennetek/gridfinity-rebuilt-openscad #68: support custom shapes other than rectangles

avxmw, 2023-02-13:
For some situations it would be nice to have baseplates and/or boxes which aren't rectangles, but L,Z,T-shaped or similar to work around corners of your storage place or not waste space on the grid building a better fitting box around your tool.

https://www.printables.com/model/313164-gridfinitris , https://www.printables.com/model/316113-non-rectangular-gridfinity-bin-parametricish-fusio

I'm not sure what the most user friendly way would be to implement that, maybe an external file with a line by line visual representation like this:

```
__###_
###_##
#____#
```

Simply screwing/gluing shapes together might work as an alternative, but then, 1x1 would be all we need to go from there :D

GreenEllipsis, 2023-03-26:
Another workaround would be to cut away what you don't need from the .stl. You could do this in PrusaSlicer with a subtractive box, for example.

Ruudjhuu, 2023-04-25:
Instead of an external file, we could create a list of list variable for this. I do not know how this should be added in the UI.

```scad
config = [
    [0,0,1,1,1,0],
    [1,1,1,0,1,1],
    [1,0,0,0,0,1]
];
```

avxmw, 2023-04-25:
I'd guess the simplest way would be to have a local web page and some JavaScript where you can "paint" squares to make your form and then generate a list like you've shown from it. I have zero knowledge about that though.

For me personally it's more important to have such a functionality at all then it being easy to use, at worst it's done trial and error and wasted CPU cycles.