kennetek/gridfinity-rebuilt-openscad #69: Bin renders as a filled box at specific z height

martinjuhasz, 2023-02-14:
Hey,

i'm using the `gridfinity-rebuilt-bins.scad` script and only changed parameters in the customizer.
Once my bin reaches a specific z height, the bin no longer renders as a bin, but basically as a solid block.
It doesn't matter which type of `gridz_define` i use, as soon as i reach this phyical height, it gets rendered as a block.
The preview is always fine, but render and stl are not.

[image]

Any ideas what might be wrong here?

P.S.: Thanks for this. This is an awesome script and i will extensively use it ;)

martinjuhasz, 2023-02-14:
So i tested some more and this seems to be related to macOS (Monterey in my case). Didn't matter if i used the `2021` or `2023` build of OpenSCAD or if `fast-csg` was enabled.

When using Windows10 with the same settings it seems fine.

kennetek, 2023-02-14:
That is very strange. I tried to replicate the issue by making a 20 unit tall bin, but it worked just fine. I am on Windows 10 using OpenSCAD 2022.11.13.ci12787 with fast csg. Sounds like Mac is the issue, thankfully (I thought I did something terribly wrong...).

martinjuhasz, 2023-02-15:
Is there any way we can debug this to make it work on macOS? Would be a bummer if this script would only run on windows although OpenSCAD is universal.

Let me know if i can log you any variables or something

Ruudjhuu, 2023-02-15:
I tried to reproduce this on a MacBook Pro:
[image]

and could not reproduce:

[image]

openscad info:
```
OpenSCAD Version: 2023.02.13 (git d7c5543dc)
System information: Mac OS X Version 12.6 (Build 21G115) MacBookPro15,1 8 CPUs 16.00 GB RAM 
User Agent: OpenSCAD/2023.02.13 (git d7c5543dc) (Mac OS X Version 12.6 (Build 21G115) MacBookPro15,1)
```

Could this issue pop up due to some setting? Maybe try to purge all settings regarding openscad from your system and reinstall? or maybe your system has a m1 processor? I don't think the issue is in the code as openscad "should" act the same on different OSs.

@kennetek for the future a tracing feature could be added so we could easily see which path the code took to check if the code is executed as expected. just a thought.

Ruudjhuu, 2023-02-15:
Of course

[image]

kennetek, 2023-02-15:
I have no clue how to fix this, and don't have time right now to figure out code checking. Maybe in the future. I don't have a Mac to test with either, so unfortunately I don't think I can help.

martinjuhasz, 2023-02-16:
Sure no problem.  We'll see if this even will happen to another person or if it's just some iffie hardware/sofware config hicckup. I'll do the Windows dance for now.

Ruudjhuu, 2023-03-18:
I added some diagnostic tracing information to this repository. @martinjuhasz, could you try to reproduce the issue again on the branch of #74, set `tracing = true` and copy the console output here (preferably an attached file). This way I can compare the program flow on your machine vs my machines and check for differences.

tobymurray, 2024-01-16:
I think I ran into a similar or related issue?
- OpenSCAD: version 2023.10.15.ai16550 (git 70309bcaa)
- gridfinity-rebuilt-openscad: e7478a7dc
- Ubuntu 22.04

Opening gridfinity-rebuilt-bins.scad, I see:
[image]

The output bin is actually solid though, not just rendered as such:
[image]