kennetek/gridfinity-rebuilt-openscad #70: Feature: Stackable lids

Ramblurr, 2023-02-17:
I would love to print some bins that I can put a lid on top of, on top of which I could stack more bins!

For my immediate usecase I wouldn't need magnets in the lid (to secure bins stacked on top), but it would be a nice bonus.

Ruudjhuu, 2023-03-17:
You mean this?

[image]

Ramblurr, 2023-03-18:
Yes, that looks very nice. Can it be parameterized to support different sizes?

Ruudjhuu, 2023-03-18:
It is a bin with `grid_z = 1`. You can set any size you want.

WizardUli, 2023-04-12:
It is not the same as having a stackable baseplate like here https://www.printables.com/model/361287-gridfinity-tray-for-ikea-samla though.

Guildenstern, 2023-12-14:
https://www.printables.com/de/model/544769-miniatures-gridfinity-holders-for-storage-and-tran this is another example. I would suggest an additional lip-style. I am currently working on an openscad clone for that miniature tray. 
Unfortunenatly my openSCAD coding style is really not sexy. If you are open for pull requests I can still create one once I am done. But it would be good when someone looksat it **in detail** before merging.

My approach is to use the base plate and cut out the "gridlines" so that I have a lip wihch I will then just add to the bin

Guildenstern, 2023-12-14:
maybe I'm just stupid. I haven't printed any bin generated with this openScad lib. But the original bins actually were stackable. The lip just looks so different than on the model I downloaded and printed before

tobymurray, 2024-03-30:
If I understand what @Ramblurr is asking for, I think it's the same kind of thing I would like. For example, I have a tower of bins and base plates in a toolbox and I don't want anything to shake around, so I want to print one more base plate that acts as a lid but still enables stackability.

A 1-height box is exactly the style of what I imagined, just tiled out instead of expanded in the same way as the base plate. For example:

### 3x3 base plate (thin style):
This is what I want on the top surface
[image]

### 3x3 1-height box:
Note no internal bins, but the feet and the solid plane on the bottom are what I'm looking for
[image]

Maybe I'm not understanding how to use the linear compartments correctly, but I would like to print 3x3 of 1-unit boxes as an entire assembly.

I exported the two objects above, sliced and stitched them together to get:
[image] which I'm hoping isn't just a silly thing to exist.