kennetek/gridfinity-rebuilt-openscad #71: Holes in 'screw together minimal' style plate are too low.

JackGM, 2023-02-17:
The holes in 'screw together minimal' style plate are too low, only being 0.5mm up from the bottom of the plate.

Ruudjhuu, 2023-02-18:
Could you share your settings with which you generate the model? When I generate it the hole is 1.53 mm up from the bottom. For an M3 screw this is enough room. Do you also have this issue with the `screw together` base? 

[image]

JackGM, 2023-02-18:
I seem to have the same problem with both the screw together, and screw together minimal.
[image]
[image]
[image]

However, after playing with the settings it seems like enabling the magnet hole seems to improve the situation. I'm not sure if this would be obvious to new users however?
[image]
[image]

JackGM, 2023-02-18:
Additionally, changing the style hole from none to countersunk, seems to drop the holes down again by about 0.45mm which means they still have a nice strong amount of material below the hole, but will print slightly faster 😄

Ruudjhuu, 2023-02-18:
@JackGM Could you try to use my branch with a possible fix if this will solve your issues stated here? 
If the fix works, I'll mark the pull request as "ready for review"

JackGM, 2023-02-18:
> @JackGM Could you try to use my branch with a possible fix if it will solve your issues stated here? If the fix works, I'll mark the pull request as "ready for review"

Hi Ruudjhuu,
Just tried your branch and that fixes the issues created with the holes being too close to the bottom. Thank you!
I am still seeing a slight change in height when 'style hole' is adjusted, but the main issue is fixed!

Ruudjhuu, 2023-02-18:
I see,

The `screw-together` feature is build on the `skeletonized` and `screw-together-minimal` is created further on `screw-together`. With the skeletonized feautre, the custom height of the bottom part of the base makes sense if switched between `none` and `any other choice`. What I do not understand is why there is also a difference between `countersink` and `counterbore`. @kennetek is this an intentional feature or a bug?

This behavior is not wanted for `screw-together-minimal` as it makes the hole obsolete and `screw-together` as it makes the different flavours incompatible.
I think there is a need to give all screw-together features a fixed height so they are compatible with each other (could be calculated with screw size and screw head as parameters). Currently we have for each `screw-together` feature 3 different heights which are kind of hidden in the `minimal` version and make them not compatible with each other.  @kennetek do you have a suggestion for a fixed / calculated height for the `screw-together` features?

My sugestion is:
```c
screw_together_offset = 0.5;
off = d_screw_head + screw_together_offset;
```
But this can be to small for `screw-together` with a screw hole style. So maybe:
```c
screw-together-minimal-height = 20; //just random for now
screw_together_offset = 0.5;
screw-together-height =  d_screw_head + screw_together_offset;
off =  screw-together-height < screw-tegether-minimal-height ? screw-tegether-height: screw-tegether-minimal-height
```

Ramblurr, 2023-02-21:
Seeing this too!

[image]

Ruudjhuu, 2023-02-22:
I created a table where `style_plate`, `enable_magnet` and `style_hole` are the input as they have influence and `off` is the output representing the height of the bottom part of the baseplate.  In my opinion this is way to complex to put in one line of code. I changed the branch so that for all screw together baseplates, the height is the same. I chose 6.75 as bottom height, as that was the larges number and I think it is needed for something as I do not know why this value is so different for every style combination of the base-plate.  This number is probably a waste of filament when you want to print the `screw-together-minimal` style. @kennetek how should we move forward?

style_plate |magnets| style_hole | off
---|---|--- | ---
0 | false | 0 | 0
0 |false |  1 | 0
0 | false | 2 | 0
0 | true | 0 | 0
0 |true |  1 | 0
0 | true | 2 | 0
1 | false | 0 | 6.4
1 | false | 1 | 6.4
1 | false | 2 | 6.4
1 | true | 0 | 6.4
1 | true | 1 | 6.4
1 | true | 2 | 6.4
2 | false | 0 | 1
2 | false | 1 | 3.5
2 | false | 2 | 4
2 | true | 0 | 3.4
2 | true | 1 | 5.9
2 | true | 2 | 6.4
3 | false | 0 | 6.75
3 | false | 1 | 5.9
3 | false | 2 | 6.4
3 | true | 0 | 6.75
3 | true | 1 | 5.9
3 | true | 2 | 6.4
4 | false | 0 | 6.75
4 | false | 1 | 5.9
4 | false | 2 | 6.4
4 | true | 0 | 6.75
4 | true | 1 | 5.9
4 | true | 2 | 6.4

geekifier, 2023-03-10:
I just wanted to chime in and confirm that @Ruudjhuu's branch fixes my issue (#79) as well. This is pretty urgent fix IMHO as basically `master` is broken at this point for anyone who wants to use magnets and/or screw together plates. 

I have been using the PR #72 as a fix, but it was not before I wasted some filament and did some head scratching trying to figure out what was going on :).

Ramblurr, 2023-03-18:
Thank you @kennetek  and @Ruudjhuu!