kennetek/gridfinity-rebuilt-openscad #77: Wrong location for magnet holes in bases?

cseelye, 2023-02-23:
I'm seeing the magnet holes in the bases are getting cut into the center of each grid instead of in the four corners. Took me  a minute to figure out why I didn't see any difference with magnets enabled/disabled, but when I change the style to the full weighted baseplate I can see the holes appear and disappear. I'm running a fresh clone of the repo.
[image]

Ruudjhuu, 2023-02-23:
You are correct, I provide a fix in #72. If you want to use this branch as a temporary workaround, watch out with using any of the screw together types as they are not final yet.

cseelye, 2023-02-23:
Awesome, thanks for the quick reply!

closed 2023-03-17 by kennetek

cseelye, 2023-04-04:
Thanks for your efforts on this, this project is fantastic and I use it all the time!