kennetek/gridfinity-rebuilt-openscad #78: Deeper bins without magnet/screw holes

tyrel, 2023-02-24:
When configuring bins, if I turn off the magnet+screw holes, there's still an ~8mm thick bottom on the bin which has no purpose. Is there an existing way to lower the internal bottom of the bin so the bottom isn't as thick? If you make the internal bottom a little smaller you could move it even further down into the base of the bin, although this could result in the walls needing to be beveled in a bit, which might be a net loss? Thoughts?

Ruudjhuu, 2023-02-25:
Duplicate of #42. 

Tried to implement it without luck in #52.

tyrel, 2023-02-25:
Thanks. I modeled it myself (without OpenSCAD) and made something that works great for a 1x1 bin, by chamfering the corners of the bottom. Here's a profile:
[image]
Unfortunately, as it seems has previously been found, this doesn't work with larger bins. You *could* have a bump up inside the bottom of each one but there are probably limited cases where that's worth it.
Instead I'm just going to make taller bins, and now playing with hollowing out the underside of the bottom to see if that reduces the amount of filament.

closed 2023-02-25 by tyrel