kennetek/gridfinity-rebuilt-openscad #81: Idea: Stacking enhancement

buginside, 2023-03-01:
Add small locking bits to prevent stacked bins from moving.
Discussed here: [https://www.reddit.com/r/gridfinity/comments/119eeqv/gridfinity_stacking_bins_that_dont_move/](url)
[image]

amalafrozalam, 2023-03-01:
I'm the OP on the Reddit post. This concept is borrowed from a lot of other assortment boxes as well. The most notable version would be the original ALCH boxes. There's also several other projects on Printables that have a similar feature.

The one in the picture is the type I've seen most executed and locks the boxes in really effectively.

amalafrozalam, 2023-03-03:
[image]

To demonstrate why this is important, here's an image of 2 bins stacked on each other. The top layer is a 2x3 next to a 2x1.

The top two bins sag together where they meet, into the bottom bin. But more importantly, you can see how the green bin isn't lining up with the red one anymore because the sagging of the red one causes it to dip just ever so slightly.