kennetek/gridfinity-rebuilt-openscad #83: How did you make a color changing render?

spuder, 2023-03-09:
I've spent a long time figuring out how to make a rotating render using openscad

https://stackoverflow.com/a/70468907/1626687
https://github.com/spuder/CAD-scripts/blob/main/stl2gif.sh

How did you make the render change colors? 

[image]

kennetek, 2023-03-13:
I don't think I still have the code snippet, it got lost somewhere and I do not know where. But I can remember vaguely how I did it. To sum it up, I used the animation window in the OpenSCAD GUI to program the movement/color, get an image dump, and used Image Magick to compile into a gif. Animating in OpenSCAD can be done with the variable `$t` (see the [wiki](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#.24t)). I think I normalized `$t` to be between 0 and 1, so `$t` was divided by the number of frames I wanted. There are three steps to this animation:
1. Rotation: use the rotate command around the Z axis. Since it only does one rotation, this is simply `360*$t`
2. Translation: The up and down movement is a sine function of `$t`. So something like `sin(360*$t)` inside a translate command. There may have been frequency modifiers. 
3. Color: using the `color[r,g,b,a]` command, all that is needed is some continuous mapping function of `$t` to a hue. I don't remember what I used, most likely something copied off Stack Overflow.

After playing the animation and having the dump images box checked, the files were in order in the script directory. With ImageMagick, you can collect all these image files and generate a gif.

spuder, 2023-03-20:
Thanks so much for the information. 

Would you have by chance used an external library like [hsvtorgb.scad](https://www.thingiverse.com/thing:279951)?

The following is close, but not quite working yet

```
use <hsvToRGB.scad>
color(hsvToRGB(240,1,1) ) import("/Users/foobar/Code/github/spuder/CAD-scripts/benchy.stl");
render();
```

Similar command from the command line (does assume the library is already installed to `/usr/local/share/openscad/libraries` )

```
    $openscad_path /dev/null \
        -D 'use <hsvToRGB.scad>;' \
        -D '$vpr = [60, 0, 360 * $t];' \
        -o "${MYTMPDIR}/foo.png"  \
        -D "color(hsvToRGB(\$t * 360,1,1)) import(\"${MYTMPDIR}/foo-centered.stl\");" \
        --imgsize=600,600 \
        --animate 60 \
        --viewall --autocenter \
        --preview \
        --quiet
```

spuder, 2023-03-20:
Thanks! 

Got it working with the `<hsvtorgb.scad>` [library](https://www.thingiverse.com/thing:279951)

It takes a value between 0.0 and 1.0 and translates it to an RGB value (e.g. [1,0,0] would be red, [0,1,0] would be green)

```
    $openscad_path /dev/null \
        -D 'use <hsvtorgb.scad>;' \
        -D '$vpr = [60, 0, 360 * $t];' \
        -o "${MYTMPDIR}/foo.png"  \
        -D "color(hsvToRGB(\$t,1,1)) import(\"${MYTMPDIR}/foo-centered.stl\");" \
        --imgsize=600,600 \
        --animate 60 \
        --viewall --autocenter \
        --preview \
        --quiet
```
        
I've created a docker container/script that can do this with any .stl file [on my github spuder/CAD-scripts](https://github.com/spuder/CAD-scripts)

Thanks again

 
[image]

closed 2023-03-20 by spuder