kennetek/gridfinity-rebuilt-openscad #85: Lite bins have weak bridges at corners

kennetek, 2023-03-17:
See #82 for more information.

rexpark, 2023-04-10:
Confirmed. I just printed a few different lite bins and they all had the issue.

rcarmo, 2023-05-07:
Same here. It's especially bad when doing 2x2 multiples, you get a single bridge layer (we should get at least another 0.2 layer on top, possibly more):

[image]

rcarmo, 2023-05-07:
Hi there! I found a fix, which was to edit [L90](https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/main/gridfinity-rebuilt-lite.scad#L90) to read:

```
translate([0,0,h_base+d_clear+0.2])
```

So I suspect `d_clear` is the thing to tweak here, but I didn't want to mess around with it lest I break something else (I know the dimensioning has a few tricks for `lite`).

rcarmo, 2023-05-07:
In the meantime, I’ve been looking at prints from before the “lite” version was reimplemented (back when I was using a fork and did a PR based on it), and I the current implementation generates much thinner bases overall - so much so that they seem flimsy when compared to the older “lite” prints. 

I looked at a GCode file I had lying around from that time and the base has almost twice the amount of layers (enough for the slicer to have generated infill), whereas the current version generates much thinner bases, so maybe we need a parameter to set the overall base thickness as well?