kennetek/gridfinity-rebuilt-openscad #90: Allow removal of specific bin sides

boc-tothefuture, 2023-03-23:
There is a useful project called [sideless bins](https://www.printables.com/model/209998-gridfinity-sideless-bins-and-platforms) which has some bins available where one or more side wall is missing, allowing you to put together bins that are bigger than your printer. 

This works well, except when you want to make custom bins.  It would be nice if there was an option to specify the sides to render.  The default would be to render all sides, but you could override that with a list of sides to render if you wanted to omit some. 

For example, right now I need to print a 2x10 bin for something and I need to have specific cutouts in that bin.  I would like to be able to print two 2x5 bins that I could place next to each other that both had one side missing.

Ruudjhuu, 2023-03-23:
I like the idea to be able to create bins bigger than your build plate. I do think the approach of sideless bins is not the way to go.

Why not generate the final object you want and split it in printable pieces. This would eliminate the calculations of where to exactly put your cutout of the two (or more) separated pieces.

I think splitting an object can be the responsibility of a slicer. The slicer already happens to know your print bed size. I did saw a neat little feature in pruca slicer alpha which we probably are not going to mimic. It can split an object in any direction and add notches for alignment when putting it together.

[Pruca slicer 2.6 features](https://www.tomshardware.com/news/new-features-added-to-prusaslicer-26-alpha-release)

boc-tothefuture, 2023-03-24:
Unrelated to sideless bins, [here](https://www.printables.com/model/432019-gridfinity-spooly-holder-big-7) is an example of a print where that removes one side of a bin.  Looking at that print, rather than a binary wall exists, doesn't exist.. A more flexible request would be to be able to set/override the height/u of each wall independently.

cseelye, 2023-04-04:
I was just coming here to request this same thing. I was also interested in one other great feature of the sideless bins project: it lets you create odd shaped bins by combining the pieces. I’m not sure how the UX would work, but it would be great to be able to create an L shaped bin or other random shapes to fit particular objects without wasting an entire NxM rectangle.

basvdijk, 2023-04-25:
Created pull request https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/104 for first setup of this feature

[image]

Ruudjhuu, 2023-04-26:
Mark this one as duplicate as #103 has more recent status updates, please discuss this feature in #103 

Duplicate of #103

Ruudjhuu, 2023-04-26:
Mark this one as duplicate as https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/103 has more recent status updates, please discuss this feature in #103

Ruudjhuu, 2023-04-26:
Duplicate of #103