kennetek/gridfinity-rebuilt-openscad #97: Dividers at non-whole "base unit" positions prevent stacked bins from nesting all the way

adalovebass, 2023-04-18:
This is present in other sizes, but this is a 1x2x3 (d,w,h) bin, with divX set to 2 and full tab. Regular lip.

Image of the cross-section, you can see the divider (and tab) (B) interfere with the two containers' angled surfaces nesting completely into each other.
[image]

Overview of the above cross-section for context
[image]

Ruudjhuu, 2023-04-25:
Thank you for submitting a issue.

I think this is by design, as the original bins from Zack are also designed this way. The offset at the sides are there to make the bins fit regardless of the accuracy of most 3d printers.

adalovebass, 2023-04-26:
That makes sense if you have divisions on whole numbers, for example, a 2x2
box with 2x2 divisions. In that case, the division walls will coincide with
the grooves in the foot of the bin. I think Gridfinity primarily contacts
by the angled walls, so if those aren't contacting due to a proud divider
wall, stacked plates will lack stability (as I'm observing).

Ruudjhuu, 2023-04-27:
I understand your stability issue. I like the the image you created to describe the problem. Maybe you can do this for the original bins: [Orignal bins](https://thangs.com/designer/models/3d-model/60721?part=364ef447-6ced-489c-9112-7e85cca4e41f) and check if my claim regarding the original design is correct.

Do you also encounter the stability issue with baseplates? If I am correct, all baseplates (also original) are designed with this "issue" as the horizontal surface (sometimes the ground, somethimes the magnets) does not have an offset but all other surfaces do. 

I myself do not encounter this problem, this might be the cause of my printer not being tuned in perfectly and printing a little bit bigger then it should. Do you know how acutrate your printer is? Is there a possibility that you print somthing of 42 mm and it turns out to be 30 (exaggerated ofcourse)?

Ruudjhuu, 2023-05-14:
Close issue as this is by design.

closed 2023-05-14 by Ruudjhuu