kennetek/gridfinity-rebuilt-openscad #98: wall thickness does not get respected on partially filled bins

avxmw, 2023-04-20:
Trying to make some partially filled boxes to cut shapes from I noticed they come out incredibly weak and I thought my printer is acting up, but it turns out, when you create a partially filled box (height internal >0, divx/y=0) all walls above the value of height internal are only a single line and not the defined d_wall (1.2 in my case).

[image]

Ruudjhuu, 2023-04-26:
Could you share the setting you used in the UI and the changes you made to the files (d_wall probably)?

I tried to reproduce this and am not able to. As you can see, my walls are nice and thick (d_wall = 3)

[image]

avxmw, 2023-04-26:
[image]

Only value changed is d_wall from 0.95 (kennetek default) to 1.2 (Zachs default).

It has something to do with the lip. With lip removed it works, with lip set to regular the wall below the lip is a single line from the height given bei internal height upwards.

Rendered with PrusaSlicer for 0.4 nozzle with 0.4 line width and 0.2 layer height.

Ruudjhuu, 2023-04-28:
Thanks for the additional information, it really helped with understanding the problem better.

For the bins of zack, the wall thickness is also 0.95. Could you elaborate a little bit more regarding the 1.2 wall thickness default? is this for a specific use case we might have missed?

So you want to create a bin with lip with a custom wall thickness, but the wall thickness `d_wall` is not respected when using a lip.

I think I understand the issue, I am going to try explain it so my tomorrow self still understands it an might be able to provide a fix. Here we go:

There are 2 wall profile generators in the code. 1 generates a simple profile (name: `profile_wall()`) when `style_lip = 0` and the other one is a more advanced profile (name: `profile_wall2()`) and used when `style_lip > 0`. 

`profile_wall()` respects the d_wall parameter as is. 

However, `profile_wall2()` also respects the `d_wall` parameter in a way where the wall thickness generated with the profile is going to be `d_wall /2`. Why is this? No idea.

When a cutter is used (`divx >0 || divy > 0`), a little bit wall is added by the cutter logic so the wall would still be the 0.95 mm as expected. 

In your case, you have `d_wall = 1.2` without a cutter (`divx = 0 && divy == 0`) which results in a wall of `1.2 / 2 = 0.6`. Depending on the slicer settings this could indeed be a single line with a line width of 0.4.

A mitigation/workaround now when no fix is available yet is to set `wall_d` twice the size as you want it to be and I expect a wall to be created with `wall_d / 2`

So if this is the issue, the bins shown in your first post inside the slicer were generated with a lip (this is not obvious from the image)?

avxmw, 2023-04-28:
Interesting, somehow I remember 1.2 and it made sense on a 0.4 nozzle.

As per your suggestion, indeed, setting d_wall = 2.4 and slicing it with the same settings results in 3 wall lines for me - so this can be a work around.

It's a bit unwieldy to be able to set the internal height in the main file, but having to change the d_wall in another. Maybe it would be better to split (semi-)filled bins out into a separate file like baseplates, lite- and spiral-mode already are?

And yes, original picture was with lip, but not shown/mentioned, sry.

gshpychka, 2024-02-19:
@Ruudjhuu is there an issue for that particular bug?

Florin-Popescu, 2024-12-25:
Hello! I'm having the exact same issue. I modified `profile_wall()` (not `profile_wall2()`!) to use `d_wall`. Walls generate correctly for me now. I'd submit a PR but I'm not sure if it wouldn't break something else and I'm not experienced with openscad.

EmperorArthur, 2025-04-19:
Fixed by #265

closed 2025-04-19 by EmperorArthur