kennetek/gridfinity-rebuilt-openscad #99: Idea: L-shaped bins e.g. for hex keys

basvdijk, 2023-04-20:
I needed some bins for my hex keys. It would be nice if I could generate from example a 4x2 L shaped bin (red) I can then make a 3x1 bin (blue) for the smaller hex-keys. These 2 bins then fit nicely together.

[image]

basvdijk, 2023-04-20:
@avxmw  I tried in Prusa slicer, but didn't got the cornering nicely

[image]

What do you mean with 2+ combined? In openSCAD?

avxmw, 2023-04-20:
You get the corners correct when you use (in your example) a 1x2 with overlap instead of a 1x1. Still need to cut the wall in between. Not a nice way, but a temporary workaround.

As with 2+ I meant 2 or more boxes - you might need a C/Z or similar shape build out of 3 or more parts.

basvdijk, 2023-04-24:
@avxmw thanks, I've tried your suggestion. Turned out that using Prusaslicer is not easy / accurate enough to get a good print. 

[image]

avxmw, 2023-04-25:
Not a pro-designer here either, personally for that task I use TinkerCAD (free account, works in browser):

1. Import 2 (or more) STLs - sadly must do them 1 by 1
2. select both, click arrange, use the little "circle" buttons as needed
3. create new shape as a cutter with x/y width - 2xwall thickness
4. use centering and movement functions to allign
5. set cutter shape to "hole"
6. select all, group
7. when happy, click export and save as one of the provided options.

Good luck.

Ruudjhuu, 2023-04-25:
Duplicate of #68

closed 2023-04-25 by Ruudjhuu