kennetek/gridfinity-rebuilt-openscad #102: License Change to MIT?

LeifMessinger, 2023-04-25:
The license for this project at the moment is CC Attribution Non-Commercial Share Alike.

During Zack Freedman's April fool's day video, [he changed the license to the MIT license](https://youtu.be/ofI2OCq_QqA?t=344), which has less restrictions. And because I can't trust an April fool's day video, [his gridfinity files on thangs have changed to MIT](https://thangs.com/designer/ZackFreedman/3d-model/Gridfinity%20Baseplates-60925), if you click the "view license" button.

Could the license for gridfinity rebuilt get changed to MIT? I know that means losing those restrictions, but it's in the spirit of the project.

LeifMessinger, 2023-04-25:
Oh, yeah sorry

closed 2023-04-25 by LeifMessinger

Ruudjhuu, 2023-04-25:
Duplicate of #96