kennetek/gridfinity-rebuilt-openscad #110: Adding text to the tabs

alex-bene, 2023-05-04:
I am not sure if this is easy to implement in openscad, but I think it would be nice if we could also add text to the tabs, which would be extruded a couple of millimeters higher.

Ruudjhuu, 2023-05-09:
I don't think it is easy to implement. It is part of the cutter logic, which I find hard to follow. 

Even if we make the text a couple mm high, it interferes  with the stack ability of the bins. I think cutting out text, or first remove 1 mm and then extrude it a mm would be better options in my opinion.

lostapathy, 2024-07-03:
I'm working on doing this and have something kinda going.

> I don't think it is easy to implement. It is part of the cutter logic, which I find hard to follow.
You're not wrong!  Extruding the label up is tricky because this would need to happen in the cutter object - which is something that's subtracted, it's not extruding new geometry.

I'm doing this by adding the text to the cutter object, which has the net effect of debossing it into the tab.  Doing it that way also keeps from having a raised label that interferes with stacking.  Then I'm using the slicer to paint a second color into the debossed text for multi-color printing.

Debossing a very minimal depth (like 0.1mm) works best for multi-color printing as it keeps the second color closer to the surface and more legible.  If you just want to print in a single color, debossing deeper makes for more readable text.

I'll try to follow up with a PR if I can get the implementation fairly clean.  So far I only really have it working for single-label bins, and I've found what I believe to be other issues with how the tabs are constructed that I want to run to ground before I submit any PRs on this topic.