kennetek/gridfinity-rebuilt-openscad #112: Changing length creates issues

mrcpu99, 2023-05-12:
Wanted to make some half-size bases and boxes for a literally little project.

However, changing length to 21 with the #3 base, puts holes in weird places, and outside the grid.  Fit to drawer appears to work, but puts holes out on the "filler" space.

The other issue was that it took seemingly forever to render.  THe regular 42 length renders in 10-15 seconds give or take, I've never actually timed it, but it's fast enough that it doesn't slow me down.  But with 21, easily 30 minutes.    And burns up 12+ GB of RAM.  And yes, fast-csg is enabled.

Not a huge big deal, more informational.  I don't know how to pull down the latest version with git, so it may be a bug that's fixed.

Ruudjhuu, 2023-05-12:
Thanks for creating an issue.

Have you tried the fix in #107? This might solve one of your issues.
The "forever render" issue is weird. Let me know if #107 solves it, if not I will look into it.

Some steps to check #107 to help you with git if you have git installed:

```sh
git clone https://github.com/Ruudjhuu/gridfinity-rebuilt-openscad.git
git checkout fix/static_hole_location
```

Ruudjhuu, 2023-05-14:
Duplicate of #95

closed 2023-05-14 by Ruudjhuu