kennetek/gridfinity-rebuilt-openscad #115: Automatically split baseplate size according to printer bed size

jottr, 2023-05-26:
I want to print a baseplate that is larger than my print bed. 
Currently I have to split the baseplate into two printable parts, that are screwed together. 
This results in wasted space, since the grid is not utilizing the available space optimally. 

Example:

[image]

It would be very useful to have two variables `print_bedx` and `print_bedy`, which allow for automatically splitting large grids.

Ruudjhuu, 2023-05-27:
I am not 100% sure if I understand your intentions. But the screw together feature is created for creating baseplates larger than print beds. If you turn of the `fit to drawer` feature and screw together multipile baseplates there wont be wasted space. 

like this:

[image]

jottr, 2023-05-27:
Thanks for trying to help.

I'm confused. So given a printer bed of 235x235 mm, and a desired base plate size of 630x294mm, how would I set this up?

I'm aware of this [section]( https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/754e555a84fce3b33127b12ca70287f129f181b6/gridfinity-rebuilt-baseplate.scad#L36C1-L40):

```
/* [Fit to Drawer] */
// minimum length of baseplate along x (leave zero to ignore, will automatically fill area if gridx is zero)
distancex = 0;
// minimum length of baseplate along y (leave zero to ignore, will automatically fill area if gridy is zero)
distancey = 0;

```

So, if I set this to my print bed size, how would I then define the desired base plate size?

I.e. 

```
/* [Fit to Drawer] */
// minimum length of baseplate along x (leave zero to ignore, will automatically fill area if gridx is zero)
distancex = 235;
// minimum length of baseplate along y (leave zero to ignore, will automatically fill area if gridy is zero)
distancey = 235;
```

merely results in this

[image]

but not this

[image]

Ruudjhuu, 2023-05-27:
I understand you want to automate it. That is currently not supported and for now it should be done manually. This would be a feature request. 

However, if you did this manually, it should not result in wasted space. 
keep in mind the fit to drawer section has 2 more variables which influence the space on the sides. 

scenario:
630x 294 will result in a grid of 15 x 7 in total. This means no fit to drawer feature is necessary: -> `distancex = 0` `distancey=0`
I would create the following baseplates with `screw_together_minimal` enabled and screw them all together or even without `screw_together_minimal` and have a perfect fit in your drawer (if you want to place it in a drawer):
 * 3 x `gridx = 5` `gridy = 5`
 * 3 x `gridx = 5` `gridy = 2`

jottr, 2023-05-27:
Thanks for the explanation. 
So let's make this a feature request then, and remove the "question" label. 
I'd be happy to help with implementation if this is a feature that would be generally desired by maintainers.

entropical, 2024-08-19:
I'd like this feature too. 

I did it manually recently, but it was messy, and I messed up some calculation by a few mm, so there is a bit of play in my drawer.

Plaethe, 2024-12-31:
I'd like this feature too due to the same messy reasons.

ewtillman, 2025-01-11:
https://gridfinity-calculator.streamlit.app/   this site will do it for anyone still looking