kennetek/gridfinity-rebuilt-openscad #116: Compatibility issues with BOSL2

koseduhemak, 2023-05-27:
Hi there,

First, thank you for the great library!
Lately I am struggling to get gridfinity to work with BOSL2... It seems that the "lip" function does not work correctly...

```openscad
include <../../libs/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad>
include <../../libs/BOSL2/std.scad>
// ===== INFORMATION ===== //
/*
 IMPORTANT: rendering will be better for analyzing the model if fast-csg is enabled. As of writing, this feature is only available in the development builds and not the official release of OpenSCAD, but it makes rendering only take a couple seconds, even for comically large bins. Enable it in Edit > Preferences > Features > fast-csg
 the magnet holes can have an extra cut in them to make it easier to print without supports
 tabs will automatically be disabled when gridz is less than 3, as the tabs take up too much space
 base functions can be found in "gridfinity-rebuilt-utility.scad"
 examples at end of file

 BIN HEIGHT
 the original gridfinity bins had the overall height defined by 7mm increments
 a bin would be 7*u millimeters tall
 the lip at the top of the bin (3.8mm) added onto this height
 The stock bins have unit heights of 2, 3, and 6:
 Z unit 2 -> 7*2 + 3.8 -> 17.8mm
 Z unit 3 -> 7*3 + 3.8 -> 24.8mm
 Z unit 6 -> 7*6 + 3.8 -> 45.8mm

https://github.com/kennetek/gridfinity-rebuilt-openscad

*/

// ===== PARAMETERS ===== //

/* [Setup Parameters] */
$fa = 8;
$fs = 0.25;

/* [General Settings] */
// number of bases along x-axis
gridx = 2;  
// number of bases along y-axis   
gridy = 3;  
// bin height. See bin height information and "gridz_define" below.  
gridz = 2;   
// base unit
length = 42;

/* [Compartments] */
// number of X Divisions (set to zero to have solid bin)
divx = 1;
// number of y Divisions (set to zero to have solid bin)
divy = 1;

/* [Height] */
// determine what the variable "gridz" applies to based on your use case
gridz_define = 0; // [0:gridz is the height of bins in units of 7mm increments - Zack's method,1:gridz is the internal height in millimeters, 2:gridz is the overall external height of the bin in millimeters]
// overrides internal block height of bin (for solid containers). Leave zero for default height. Units: mm
height_internal = 0; 
// snap gridz height to nearest 7mm increment
enable_zsnap = false;

/* [Features] */
// the type of tabs
style_tab = 5; //[0:Full,1:Auto,2:Left,3:Center,4:Right,5:None]
// how should the top lip act
style_lip = 0; //[0: Regular lip, 1:remove lip subtractively, 2: remove lip and retain height]
// scoop weight percentage. 0 disables scoop, 1 is regular scoop. Any real number will scale the scoop. 
scoop = 0; //[0:0.1:1]
// only cut magnet/screw holes at the corners of the bin to save uneccesary print time
style_corners = false;

/* [Base] */
style_hole = 3; // [0:no holes, 1:magnet holes only, 2: magnet and screw holes - no printable slit, 3: magnet and screw holes - printable slit]
// number of divisions per 1 unit of base along the X axis. (default 1, only use integers. 0 means automatically guess the right division)
div_base_x = 0;
// number of divisions per 1 unit of base along the Y axis. (default 1, only use integers. 0 means automatically guess the right division)
div_base_y = 0; 

// ===== IMPLEMENTATION ===== //
puffer = 3.75;
drillsBeton = [
    [8, 120],
    [6, 100],
    [5, 85],
    [4, 75]
];
extractorLengthBeton = 30;

drillsSteel = [
    [6, 93],
    [5.5, 93],
    [5, 86],
    [4.5, 80],
    [4, 75],
    [3.5, 70],
    [3, 61],
    [2.5, 57],
    [2, 49],
    [1.5, 40],
    [1, 34],
];
extractorLengthSteel = 20;

drillsWood = [
[10, 120],
[8, 117],
[6, 93],
[5, 86]
];
extractorLengthWood = 30;

height = 10;
extractSpace = 0;

gridfinityInit(gridx,gridy,height(gridz), 0, length) {
    cut_move(x=0, y=0, w=.21, h=3) {
        drillHolder(drillsSteel, extractorLengthBeton, extractSpace);
    }
    
    cut_move(x=gridx-1, y=gridy-1, w=length/gridx/100, h=3) {
    translate([23,20,0]) {
        rotate([0,0,270]) {
         text3d("Stahl", h=2, size=8);
         }
         }
    }
}

gridfinityBase(gridx, gridy, length, 0, 0, 1);

module drillHolder(drills, extractorLength, extractSpace) {
    for (i = [0:len(drills)-1]) {
        fromLeft = drillPlacement(drills, 0, i, puffer);
       
        toBottom = i==0 ? 0 : drills[0][1]/2-drills[i][1]/2;
           translate([fromLeft,-toBottom,0]) {
            dim = [drills[i][0], drills[i][1]+extractSpace, drills[i][0]*2];
            cuboid(dim, rounding = drills[i][0]/2);
            
            rotate([0,0,270]) {            
                translate([-10, drills[i][0]/2+.5, 0]) {

                    text3d(str(drills[i][0]), h=2, size=3);
                }
           }
           }
           
           translate([fromLeft,-drills[0][1]/2,0]) {
            cuboid([drills[i][0]+2,extractorLengthBeton, drills[i][0]*2], rounding = drills[i][0]/2);
            
            }
          
    }
   
}

function drillPlacement(drills, i, count, puffer) = 
(
    i<count ? (drills[i][0]+drillPlacement(drills, i+1, count, puffer)+puffer) : drills[i][0] / len(drills)
);

```

If I want to generate the lip using `style_lip=0`, I get the following error:
```
ERROR: Assertion '_valid_line([p1, p2], dim = 2)' failed: "Invalid line." in file ../../libs/BOSL2/geometry.scad, line 235
TRACE: called by 'line_normal' in file ../../libs/BOSL2/attachments.scad, line 2787
TRACE: called by 'point3d' in file ../../libs/BOSL2/attachments.scad, line 2787
TRACE: called by '_find_anchor' in file ../../libs/BOSL2/attachments.scad, line 2537
TRACE: called by '_attach_transform' in file ../../libs/BOSL2/attachments.scad, line 2015
TRACE: assignment to 'm' in file ../../libs/BOSL2/attachments.scad, line 2015
TRACE: called by 'attachable' in file ../../libs/BOSL2/shapes2d.scad, line 67
TRACE: call of 'square(size = [0.95, 0], center = undef, anchor = [-1, -1], spin = undef)' in file ../../libs/BOSL2/shapes2d.scad, line 64
TRACE: called by 'square' in file ../../libs/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad, line 177
TRACE: called by 'difference' in file ../../libs/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad, line 170
TRACE: call of 'profile_wall_sub()' in file ../../libs/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad, line 169
TRACE: called by 'profile_wall_sub' in file ../../libs/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad, line 185
TRACE: called by 'difference' in file ../../libs/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad, line 184
```

Using `style_lip=1` everything works fine:
[image]

Any ideas how I can mitigate that?

Thank you!
koseduhemak

Ruudjhuu, 2023-05-27:
Well openscad square function accepts 0 as height. Which does not make sense at all as it does not create any kind of 2d or 3d object. So BOSL2 is correct to not accept this nonsense. I created a pull request which should solve your issues.

Ruudjhuu, 2023-05-27:
Tested your code:

[image]

koseduhemak, 2023-05-28:
Thank you very much!
Didn't thought that someone responds in such a quick manner! In the meantime I played around with the old BOSL (1) library, which worked as well. But BOSL2 has many more feature - so really nice that you could fix the problem :).

PR #117 (2023-05-29): remove square with zero height call
fixes #116 

Current code base contains a `square` call with height= 0. This results in nothing. Opensad is fine with it unless BOSL2 is used. I removed the call.
closed 2023-05-29 by Ruudjhuu