kennetek/gridfinity-rebuilt-openscad #125: How to use

joshcomley, 2023-07-30:
Before I continue:

- I'm relatively new to OpenSCAD but have done quite a lot
- I've read the docs

How am I supposed to use this to make my custom prints?

If I import the `.scad` files from another `.scad` file, say I import `gridfinity-rebuilt-baseplate.scad` into my own `.scad` file, then the "implementation" section renders it out before I can tweak it. I can't modify the variables in my own `.scad` file. The only option seems to be to call it from the command line with the custom variables, which makes mathematical things tricky.

I would have expected something like this:

    include <gridfinity-rebuilt-openscad\gridfinity-rebuilt-baseplate.scad>
    
    gridfinityBaseplate(...);

If I modify the files in the git repo to get what I want, then pulling the latest changes becomes a problem - same if I just copy all the complex OpenSCAD code into my own files.

To me it seems the modules and the default variables should be separated from another scad file that just imports the two and performs a render.

Unless I'm fundamentally missing something?

adrianVmariano, 2023-10-07:
I can say that I'm an expert user who doesn't generally remember that the customizer exists :) and I started by working directly with the code, importing the "bins" file it with `use` and setting up a module invocation to produce a bin, and it's not very friendly for direct use this way.  The friendly parameters used by the customizer have cryptic abbreviations when calling the modules, and they lack (documented) defaults, but in particular, there is reliance on parameter passing through global variables (e.g. style_lip), so you cannot just `use` or `include` that file without having to modify it.  If you want to use gridfinity-rebuilt-utility then you must use `include` because there is no way to set the required global parameter(s) with `use`.  If global variable parameter passing is the strategy of choice, it would be better to use $ variables so that they can be set from outside.   But generally I'd favor an API that does not use global variable parameter passing, but instead allows parts to be created by calls into the library with regular parameters.  (If global $ variables are needed internally this can be hidden from the user.)  

It's also unclear why the base call is a separate call.    Does that enable some capability?

Ruudjhuu, 2023-10-25:
I do fully agree with the global variables. I tried multiple times to get rid of it, but they are a big pain in the ass. Any pull request regarding improvements here are welcome. (also due to no proper tests (and this is very difficult in openscad), I am scared to touch some parts of the code).

Regarding the separate base call is to let the power-users decides for themselves what they want to put on the base. It does not have to be a bin with walls. Could also be a figure of Yoda if they like.