kennetek/gridfinity-rebuilt-openscad #126: lite bins connecting base is only one layer thick at 0.28mm

rcarmo, 2023-08-26:
Very much like in #36, there are still a few problems with lite bins. This particular connection between bases is too thin:

[image]

Is there a way to set the thickness for these connection elements? Otherwise this just makes the bins fragile (and prone to holes in these regions).

rcarmo, 2023-08-26:
Here's an example of what happens with lateral compression - those zones are the centerpoint for cracks in the print. It doesn't crack in because the spar holds it in, but the layers around those corner joints are too thin:

[image]

This leads me to believe that the entire base structure needs to be a bit thicker indeed.

WizardUli, 2023-09-10:
And for me there is no connection at all even when sliced on 0.2mm profile.
[image]

ch3vr0n5, 2023-10-18:
I have never wrestled with OpenSCAD before but I managed to fix this a little bit. Not the most elegant solution, to be sure. It creates another slim section of base and merges it with the existing one to add fillets back around the wall joints and another layer through the base connecting lines.

`module gridfinityLite(gridx, gridy, gridz, gridz_define, style_lip, enable_zsnap, length, div_base_x, div_base_y, style_hole, only_corners) { 
    union() {
        difference() {
            union() {
                gridfinityInit(gridx, gridy, height(gridz, gridz_define, style_lip, enable_zsnap), 0, length)
                children();
                gridfinityBase(gridx, gridy, length, div_base_x, div_base_y, style_hole, only_corners=only_corners);
            }

            difference() {
                union() {
                    intersection() {
                        difference() {
                            gridfinityBase(gridx, gridy, length, div_base_x, div_base_y, style_hole, -d_wall*2, false, only_corners=only_corners);
                            translate([-gridx*length/2,-gridy*length/2,2*h_base])
                            cube([gridx*length,gridy*length,1000]);
                        }
                        translate([0,0,-1])
                        rounded_rectangle(gridx*length-0.5005-d_wall*2, gridy*length-0.5005-d_wall*2, 1000, r_f2);
                        translate([0,0,bottom_layer])
                        rounded_rectangle(gridx*1000, gridy*1000, 1000, r_f2);
                    }
                    translate([0,0,h_base+d_clear])
                    rounded_rectangle(gridx*length-0.5005-d_wall*2, gridy*length-0.5005-d_wall*2, h_base, r_f2);
                }

                translate([0,0,-4*h_base])
                gridfinityInit(gridx, gridy, height(20,0), 0, length)
                children();
            }

        }
        difference() {
            translate([0,0,-1.6])
                difference() {
                    difference() {
                        union() {

                            gridfinityInit(gridx, gridy, height(gridz, gridz_define, style_lip, enable_zsnap), 0, length)
                            children();
                        }

                        difference() {

                                intersection() {
                                    difference() {
                                        gridfinityBase(gridx, gridy, length, div_base_x, div_base_y, style_hole, -d_wall*2, false, only_corners=only_corners);
                                        translate([-gridx*length/2,-gridy*length/2,2*h_base])
                                        cube([gridx*length,gridy*length,1000]);
                                    }
                                    translate([0,0,-1])
                                    rounded_rectangle(gridx*length-0.5005-d_wall*2, gridy*length-0.5005-d_wall*2, 1000, r_f2);
                                    translate([0,0,bottom_layer])
                                    rounded_rectangle(gridx*1000, gridy*1000, 1000, r_f2);
                                }

                            translate([0,0,-4*h_base])
                            gridfinityInit(gridx, gridy, height(20,0), 0, length)
                            children();
                        }

                    }
                    translate([0,0,9])
                    rounded_rectangle(gridx*1000, gridy*1000, gridz*1000, gridz);
                }
                    translate([0,0,0])
                    rounded_rectangle(gridx*1000, gridy*1000, 5, r_f2);
            }
    
    }
}`

lowell80, 2023-11-11:
@WizardUli, I found that the original design requires slicing at 0.15 (or lower, presumably) to get more than one connecting layer.  (And I too found a few option where there was no bridging).    Lowering the layer height, of course, directly contradict my primary use case of fast-printing bins.  I image that's true of others as well.

I'm printing at 0.6mm nozzle with 0.35 layer height, and can confirm that @ch3vr0n5 fix gets the job done at that layer height.  (Exact profile [here](https://github.com/lowell80/gridfinity-rebuilt-openscad/blob/dcd2000de0fb7489d2c6f52f404922ad15591908/profile_pla_n06.ini) if interested).  I did notice that there's a subtle difference (other than additional layer heights) where theres a bit more material on the outside edges where the wall meets the top surface of the connecting bins.   I don't suspect this will be a problem, or do I know if this is intentional or not.   I'd guess this would make the parts even stronger (just a guess).

Here's a slicer example of the original model (left) and the proposed fix from ch3vr0n5 on the right.  The arrow highlights an example of the difference I noted above.

[image]

Thanks a ton @ch3vr0n5 for the improvement, I've added it to my branch for now; from what I'm seeing so far, it seems like this would be a great addition to the official version.  And thanks to @kennetek for making this available in the first place.   I'm super new to OpenSCAD but I'm appreciating the ability to script this and generate tons of bin/base combos with minimal effort.  (I've done this from Fusion360 in the past, and this is way easier.). Thanks!

I'm going to print some of these and see how it works in practice!

lowell80, 2023-11-13:
Unfortunately, at 0.35 layer height there will still issues with some small portions of the connecting layer.  I re-printed with a 0.30mm layer height it was much better, but still weak in a few points.  I also noticed some artifacts where the additional thickness doesn't align quite right with the rest of the model.  So it doesn't seem like the changes from @ch3vr0n5 fixes fully address the issue, but it is a step in the right direction.  But, for anyone that needs to print one right now, before an official fix, this is a good option.

I can post some picture if helpful.

closed 2024-01-06 by Ruudjhuu