kennetek/gridfinity-rebuilt-openscad #130: WARNING: No top level geometry to render

cambevis, 2023-10-05:
These used to work for me up until recently.  I  updated OpenSCAD to 2023.09.30, and I think that's when it stopped working.

All renders return: 
WARNING: No top level geometry to render

CullenJWebb, 2023-10-07:
Same issue here, downgrading for now.

I don't see any warning or deprecation messages, so I am suspicious of OpenSCAD on this one.

killerbee84, 2023-11-07:
Can someone let me know which version they are using and it's working. Tried a couple of them but none worked. Same "no top level geometry to render".

spirilis, 2023-12-22:
Ditto with OpenSCAD dev snapshot 2023.12.20

kennetek, 2023-12-22:
Can anyone provide more information about this? On my machine (Windows 10) I am able to generate a regular bin with the December 20 build of OpenSCAD. 

Normally when "No top level geometry to render" is sent to the console its because the main module has been commented out (ie gridfinityInit or gridfinityBase)

Klaws--, 2024-10-21:
Example for code which fails:

```
$fn = 256;
d_leitung = 4.1;
d_loch = 24.0;
t_loch = 40;
d_kragen = d_loch + 10;
t_kragen = 5;
difference() {
    union() {
        cylinder(d=d_loch, h=t_loch, $fn=6);
        scale([1, 1, t_kragen/d_kragen*2]) intersection() {
            sphere(d=d_kragen);
            translate([-d_kragen/2, -d_kragen/2, -d_kragen]) cube([d_kragen, d_kragen, d_kragen]);
        }
    }
    translate([0, 0, -d_kragen]) cylinder(d=d_leitung, h=t_loch+1+d_kragen);
    translate([-50, 0, -10]) cube([100, 100, 100]);
}
```

Example for code which works:

```
$fn = 256;
d_leitung = 4.1;
d_loch = 24.0;
t_loch = 40;
d_kragen = d_loch + 10;
t_kragen = 5;
difference() {
    union() {
        cylinder(d=d_loch, h=t_loch, $fn=6);
        scale([1, 1, t_kragen/d_kragen*2]) difference() {
            sphere(d=d_kragen);
            translate([-d_kragen/2, -d_kragen/2, 0]) cube([d_kragen, d_kragen, d_kragen]);
        }
    }
    translate([0, 0, -d_kragen]) cylinder(d=d_leitung, h=t_loch+1+d_kragen);
    translate([-50, 0, -10]) cube([100, 100, 100]);
}
```

The now failing version worked before I updated to the 2021.01 version (previous version in use was 2019.5 IIRC). 

Both code examples _should_ create the same result; the difference is that the failing version uses intersection() in line 9, while the working version uses difference() instead (and adjusted coordinates for the cube to make up for the difference between intersection() and difference()).

Everything works in preview, just render throws the error message.

Further observations: when I remove the $fn=6 in the first cylinder, the failing code no longer throws an error. However, the rendered result is wrong! The "mushroom cap" at the bottom is missing in the render result (but is shown in preview).