kennetek/gridfinity-rebuilt-openscad #133: Change scoop radius

jrj2211, 2023-10-15:
I'd like to be able to change the scoop radius so that it takes up less space in the bin. Right now it takes up about 75% of the width of a 1x1 bin, I'd like to change that to 50% or so.

Been playing around with the code in `module profile_cutter` which seems to affect it, but cant quiet figure out how to get it to work. Seems like this little bit is what is cutting the radius, otherwise its just a steep ramp:

```
intersection() {
    circle(scoop / 1.1);
    mirror([0,1]) square(2*scoop);
}
```

jrj2211, 2023-10-15:
```
scoop = s ? (length*(($dh-2)/7+1)/12-r_f2) : 0; 
```

Appears that changing this line can shrink or enlarge the scoop by changing the `/12'. Making it larger shrinks the radius, making it smaller increases the radius.

I've changed mine to this which makes the scoop radius about half the length.

```
scoop = s ? (length*(($dh-2)/7+1)/15-r_f2) : 0; 
```

Would be nice to have some easier to understand constant for changing this.

kennetek, 2023-10-15:
Are you using the latest version of the scripts? It was changed a while ago such that the scoop value input would be a real number, so the scoop would be a default size (Zack's size) at 1, and you could input a value of 0.5 for it to be half that, or 0 to disable it.

jrj2211, 2023-10-15:
I thought i had as I just downloaded from printables for the first time last night while getting into this. I see now that that was an upload date of Dec 2022 and see in this repo there is the scoop variable. Thanks for letting me know this is already in here!

closed 2023-10-15 by jrj2211

kennetek, 2023-10-15:
Yeah I am sorry, the Printables files are nearly always out of date

jrj2211, 2023-10-15:
No worries, cloning from the github repo is definately the way to go