kennetek/gridfinity-rebuilt-openscad #134: utility for curved cuts

sgraham, 2023-10-17:
Hi, thank you for building this, it's really nice! (And thank you so much for the note about "fast-csg", I was going crazy waiting.)

I was trying to make a cutlery holder, and I was wondering if it'd be possible to use the existing functionality in bins to cut "nicer" curves when using `cut_move()`.

In particular, instead of my ugly square-edged cube cutouts as below, I was trying to figure out how to round those corners (examples circled in red) to make them match the typical bin curve. (I played with `fillet_cutter` and some other stuff in utility, but didn't get too far.)

```
difference()
{
gridfinityInit(4, 5, height(10), 0, 42) {   
    cut(x=0, y=0, w=.9, h=2, t=5, s=0); // big fork
    cut(x=0, y=3, w=.9, h=2, t=5, s=0); // big fork
    cut_move(x=0, y=2, w=.9, h=1)
        cube([12, 50, height(18)], center=true);
    
    cut(x=.9, y=0, w=1.3, h=2, t=5, s=0); // big spoon
    cut(x=.9, y=3, w=1.3, h=2, t=5, s=0); // big spoon
    cut_move(x=.9, y=2, w=1.3, h=1) 
        cube([12, 50, height(18)], center=true);
	
    cut(x=2.2, y=0, w=.8, h=2, t=5, s=0); // small fork
    cut(x=2.2, y=2.5, w=.8, h=1.6, t=5, s=0); // small fork
    cut_move(x=2.2, y=1.5, w=.8, h=1) 
        cube([12, 50, height(18)], center=true);

    cut(x=3, y=0, w=1.0, h=2, t=5, s=0); // small spoon
    cut(x=3, y=2.5, w=1.0, h=1.4, t=5, s=0); // small spoon
    cut_move(x=3, y=1.5, w=1.0, h=1) 
        cube([12, 50, height(18)], center=true);
}
translate([-500, -42*2.8925, 28]) cube([1000, 100, 100]);
}
gridfinityBase(4, 5, 42, 0, 0, 1);
```
[image]

Ruudjhuu, 2023-10-25:
I think you are suggesting to make a cutter with negative corner fillets. I am not sure how to do this. It will probably result in a whole new cutter type. It is also a very specific usecase. You could give it a try, pull requests are welcome.

sgraham, 2023-10-28:
OK, thanks for looking! I didn't figure it out either. :)

closed 2023-10-28 by sgraham

mrcpu99, 2023-10-29:
Did you by chance look at the openscad-fillets project on github?  Seems
like it might do what you need.

https://github.com/ademuri/openscad-fillets