kennetek/gridfinity-rebuilt-openscad #136: Skeleton with no magnets or screws too thick

jrj2211, 2023-10-18:
I'd like a skeleton base plate no screws or magnets, just to give it a little more rigidity and thickness. 

```
style_plate = 2; // [0: thin, 1:weighted, 2:skeletonized, 3: screw together, 4: screw together minimal]
enable_magnet = false; 
style_hole = 0; // [0:none, 1:contersink, 2:counterbore]
```

Given the following settings, Id expect the skeletonized base to just be the height of the `h_skel` parameter. It does not though end ends up super thick for no reason.

I believe this code should be changed:

```
function calculate_off(sp, sm, sh) = 
    ...
                    : 0)+(sh==0
                        ? d_screw
                        : sh==1
                            ?d_cs
                            :h_cb);
```

To this: 
```
function calculate_off(sp, sm, sh) = 
    ...
                    : 0)+(sh==0
                        ? 0
                        : sh==1
                            ?d_cs
                            :h_cb);
```

To me, it doesn't make sense why the height of the baseplate would add on the diameter of the screw, especially with `style_hole` set to `none`. It should just fall back to the `h_skel` height set.

Ruudjhuu, 2023-10-25:
Thanks for the proposal.

So you want the baseplate to be more ridged, but currently it is too ridged for your liking? I am not sure why you want to create a baseplate with more strength. Could you explain to me how you use the baseplates. If a baseplate is located on a flat surface (desk or drawer) then there is no need for rigidity because the flat surface should give enough rigidity. In my opinion you are adding more plastic for nothing which is a red flag for a good design. If you have a different use case, iThe best way forward might be to introduce a new baseplate type.

Nonetheless, because I think there is no real use case for your request, I am fine with making the height of an empty skeletonized baseplate configurable. You can create a pull request.

jrj2211, 2023-10-27:
I was more so just playing around with the variables, trying to figure out what I want for my baseplates. I'm not doing magnets, and I just noticed that nothing seemed to affect the skeleton height with magnets removed. My super thin example was just to make it clear the dimension I was trying to modify. Really I was just trying to make it a bit thinner than it is currently, like 3.35mm down to 2mm just to save a little on filament. But I'm just getting into gridfinity and I don't really have a use case for this so I'll close the issue. I plan on putting these in a drawer, so as you pointed out there's no reason to make them more rigid.

closed 2023-10-27 by jrj2211