kennetek/gridfinity-rebuilt-openscad #139: Baseplates are 0.5mm short per dimension

namore, 2023-10-25:
As discussed [here](https://www.reddit.com/r/gridfinity/comments/17fj1pz/comment/k6d7fkw/?utm_source=share&utm_medium=web2x&context=3), the Gridfinity Rebuilt base plates are 0.5mm shorter than other base plates and shorter than specified by Zack.

Bins have a 0.25mm tolerance at each of their four sides so they wont bump into each other, which gives the 0.5mm per dimension. The baseplates should not have this tolerance, however. I suspect the gap was added to the baseplates, as well, by accident.

The faulty code I suspect is:
[image]

charely6, 2023-10-26:
I understand its to add tolerances but it should be written in the constants as a tolerance offset so it can easily be changed for people who don't want it and it  should also be double checked for consistency.

 I know this is technically from a project using the rebuilt-utility (and an older version) but I am seeing places where the top box ends up a different size compared to the bottom box due to the tolerance subtracting being inconsistent. 

https://www.printables.com/model/602132-parameterized-labeled-socket-organizers-for-multic

[image]

t-makaro, 2023-10-29:
I have noticed this as well. This definitely should at least be customizable. My printer's tolerance for this is easily less than 0.1mm.

schorsch3000, 2024-01-04:
is that the reason why one 2x1 baseplate is 0.8mm wider than two 1x1 baseplates side by side?

closed 2024-01-06 by Ruudjhuu

namore, 2024-02-24:
Sorry for my late reply I had disable notifications apparently. Thanks for making this configurable!

For the record: My main issue with the 0.5mm tolerance was that different designed Gridfinity baseplates (that don't have that tolerance) didn't align properly with the rebuilt ones. But I also believe that the tolerance is problematic if your plates cannot move freely, e.g., due to spacers or glued together baseplates.

Thanks again for the fix!