kennetek/gridfinity-rebuilt-openscad #145: Magnet slots on bottom of bin are too small

dkovar, 2024-01-16:
The no-glue slots on the bottom of the bins appears to be 5.8mm wide. The magnets are 6mm wide. I cannot insert the magnets without forcing them and that usually results in deforming the slot. 

Am I failing to understand how to use these?

All other dimensions appear to be correct.

Ruudjhuu, 2024-01-16:
This is the idea of the no glue holes, it is press fit. So you should force it so they won't fall out. If you measure the printer hole, do you get 5.8mm ?

dkovar, 2024-01-17:
Greetings,

There is an option to turn on slits rather than holes. If you use this
option, it prints with a solid bottom and a slit in the side to insert the
magnet. The magnet ends up in the same location but with a thin layer below
it to hold it in place. This is different than the press-fit approach.

The problem is that the slit is too small to force the magnet into the slot.

-David

MLCXXX, 2024-02-18:
Hello, 
the deformation limits the stacking capability and the overall height no longer corresponds to the standard dimensions. An increase around the z-axis of the cut-out while maintaining the dimensions in the X/Y-axis should provide sufficient support for the magnets or, if necessary, a "ribbed" wall on the X/Y-axis. Perhaps that would be a solution?

jpypi, 2024-04-11:
So, I believe @dkovar is referring to the `style_hole = 4` option, which is the "Gridfinity Refined hole".
The standards file has the magnet radius (r_hole2) set to: `3.25mm`, which gives a diameter of `6.5mm` but then in the `refined_hole()` module, `r_hole2` is subtracted by `0.32mm` to reach a diameter of `5.86mm`.
@dkovar You could try adjusting the `r_hole2` variable in `standard.scad`? Maybe `r_hole2 = 3.28`?

I would think `0.14mm` should be ok for a press fit, though maybe that needs to be reconsidered?

marcodejongh, 2024-10-29:
I've also been struggling to press the magnet in on the side, and I've noticed it also makes the bottom bulge is this expected?

[image]

WizardUli, 2025-01-06:
> I've also been struggling to press the magnet in on the side, and I've noticed it also makes the bottom bulge is this expected?
> 
> ![BB1C30C4-AE3E-485F-A44B-4AAFCADB0917](https://private-user-images.githubusercontent.com/1107647/381096940-61c151f1-d6fb-4094-a686-f482472645f8.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzY5MzIsIm5iZiI6MTczNjE3NjYzMiwicGF0aCI6Ii8xMTA3NjQ3LzM4MTA5Njk0MC02MWMxNTFmMS1kNmZiLTQwOTQtYTY4Ni1mNDgyNDcyNjQ1ZjguanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA2VDE1MTcxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhYTBiYmUyMGQzYzIzZjI3MDBmMjg4ZGM4OTJmYzJiMDQ0ZGYwNmRmNTFiMDJkZmE2OTk1N2Q4MTRmNGJjNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GLuRBPz-NgFEJX9mdDzn9OVBnJb3ogqt4rVmptwuNRQ)

I've just printed my first refined-hole bin (default settings) and it came out exactly same as on your picture. I measured my magnets using my Mitutoyo calipers to be 1.99mm x 6.01mm which considering +-0.02 measurement error indicates really 2x6 size. I've measured yet unfitted holes to be 5.8mm dia and 1.87mm height quite consistently.

Perhaps other people are using some undersized chinesium from Ali?

esamecar, 2026-01-19:
I encounter exactly the same problem. If I would use only to magnets the bin would not stand flat on a surface. According to my calipers my magnets are perfect in size. The height is the bigger problem than the width.