kennetek/gridfinity-rebuilt-openscad #147: Bins are solid (filled in)

tobymurray, 2024-01-16:
Opening gridfinity-rebuilt-bins.scad in OpenSCAD I get a solid bin.

What I see when I open the file:
[image]

If I render it, I get pretty much the same thing:
[image]

Exporting the STL just to make sure it's not something quirky in OpenSCAD:
[image]

Versions:
- OpenSCAD: version 2023.10.15.ai16550 (git 70309bcaa)
  - also tried OpenSCAD version 2024.01.14.ai18102 (git ffb0152f1)
- gridfinity-rebuilt-openscad: e7478a7dc
- Ubuntu 22.04

jasper-tms, 2024-01-16:
On the right hand side set `divx` and `divy` to nonzero values. (It says in those sections `set to zero to have solid bin` so that is the intended behavior – though I do find it a bit unintuitive that the default values upon opening this file are to have these as 0 instead of 1.)

closed 2024-01-16 by Ruudjhuu

tobymurray, 2024-01-17:
> I do find it a bit unintuitive that the default values upon opening this file are to have these as 0 instead of 1

Agreed - I think I may have actually run into this before and been more patient last time and read through all the options. I proposed #148 as I think that's a value that fits closer to expectations. :shrug: