kennetek/gridfinity-rebuilt-openscad #151: Licensing inconsisency

smkent, 2024-01-24:
[As documented in this repository's `README.md`](https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/e7478a7dc9d6556dace9582d78e22a9b938e4713/README.md), gridfinity-rebuilt-openscad is MIT licensed (along with Gridfinity itself). [The Printables page for this project](https://www.printables.com/model/274917-gridfinity-rebuilt-in-openscad) lists the license as Creative Commons Attribution—Noncommercial—Share Alike. 

I know Printables does not include the MIT license as an option when publishing a model, so an alternative needed to be chosen. However, the two licenses have different restrictions (e.g. no commercial use in the published project on Printables).

I'm getting ready to publish [a new storage box model that uses gridfinity-rebuilt-openscad as a library](https://github.com/smkent/monoscad/tree/main/gridfinity/rugged-box). I'd like to license my model under an MIT compatible license for publishing on Printables. Currently if I publish a remix of gridfinity-rebuilt-openscad on Printables, Printables will (correctly) restrict me from choosing a more permissive license than the current noncommercial license.

Would it be possible to update the project's license on Printables (to Attribution—Share Alike, for example) and/or clarify the intended licensing structure?

gridfinity-rebuilt-openscad has been incredibly useful for me, both in making numerous custom bins and adding Gridfinity compatibility to other models. Clarifying the license structure would make it easier to publish remixes and usages of this project.

Thanks so much!

kennetek, 2024-01-24:
Gridfinity used to be Creative Commons Attribution—Noncommercial—Share Alike, so that's why it was that way on Printables. I had forgotten to update Printables when the original's license was updated, and I have now updated on Printables to just Share Alike as you suggested. Ideally, it would be MIT but yeah, that's not an option. 

If anyone reading this in the future has any confusion, my intention is that the license should be the same as Zack Freedman's Gridfinity license.

smkent, 2024-01-24:
Thank you, cheers!

closed 2024-01-24 by smkent