kennetek/gridfinity-rebuilt-openscad #152: Default values of gridfinity-rebuilt-bins (in particular height_internal = 0) produce filled bin with no tab

JacquesOpaux, 2024-01-24:
I'm afraid I'm missing something very obvious, but gridfinity-rebuilt-bins does not seem to work as intended. More specifically :
    - setting height_internal = 0 results in a "full" bin
    - I cannot get tabs

To check that it is not specific to my installation, I downloaded the latest stable version of openscad, the current gridfinity-rebuilt repository, opened gridfinity-rebuilt-bins.scad and rendered with default values. This produced the picture below.
[image]
I also checked on another computer with the same result.

An older version of gridfinity-rebuilt works as intended.

For what is worth, a negative value for height_internal does produce some depth in the bin.

colde, 2024-01-25:
Setting the divx and divy to 1 should get you what you are looking for. The default is no compartments, but divx and divy controls the number of compartments.

[image]

This the place to adjust it if using the customizer.

JacquesOpaux, 2024-02-01:
Thank you very much for taking the time. And indeed, I was missing something very obvious...

closed 2024-02-01 by JacquesOpaux