kennetek/gridfinity-rebuilt-openscad #153: Customisable Tab size

phillipthelen, 2024-01-26:
Currently the tabs have a fixed size of 15.85mm deep and 42mm width (if not set to full width). When generating bins that have 1x1 trays, the tabs take up a lot of the surface area making it harder to reach into the bin. For using slimmer labels, this could be changed to a customisable variable to save filament and make smaller trays more open.

If people agree this would be a useful setting to have, I can work on implementing it. Since it already is a static variable in standard.scad, it shouldn't be too hard making a custom option.

Asger1002, 2024-02-03:
I've added this in my pull request at https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/160

BenBergman, 2024-02-27:
That PR has been merged but seems to only work for bins created using the OpenSCAD modules, not with the main bin generator (at least not that I could find). An update that exposes the tab sizes for the main bin generator would be great.

EmperorArthur, 2025-06-26:
#305 will support custom bins with custom tab placement and width.  However, tab depth is fixed, and does not expose the option to the customizer.

Unfortunately, adding every option to the customizer would make it incredibly difficult to use.  I could see an automatic depth function though.