kennetek/gridfinity-rebuilt-openscad #154: ERROR: Rendering cancelled by exception boost::bad_any_cast: failed conversion using boost::any_cast

ctzurcanu, 2024-01-26:
Changed:

// length of a grid unit // originally 42
l_grid = 45;

Rendering:

Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
WARNING: [fast-csg] Corefinement corefinement mesh difference #21 failed with an error: CGAL ERROR: assertion violation!
Expr: nodes.size() == std::set<Point_3>(nodes.begin(), nodes.end()).size()
File: /Users/distiller/libraries/install/include/CGAL/Polygon_mesh_processing/internal/Corefinement/intersection_nodes.h
Line: 440

ERROR: Rendering cancelled by exception boost::bad_any_cast: failed conversion using boost::any_cast
WARNING: No top level geometry to render

aderusha, 2024-02-27:
This error also appears on the model `gridfinity-rebuilt-baseplate.scad` when setting `style_plate =` to 2 or 3, no other values changed.

starthal, 2024-03-18:
@aderusha 
Workaround for the baseplate: 
Change line 191 from
```
        circle(r_skel);
```

to
```
        circle(r_skel / 2);
```

That circle seems to describe this fillet:
[image]

If the radius is `r_skel`, the cutout is flush with the grid square:
[image]

And Something Bad happens... I'm not sure what.