kennetek/gridfinity-rebuilt-openscad #163: Skeletonized baseplate with magnets is broken

evranch, 2024-02-06:
Running OpenSCAD 2024.01.26.ai18259 (latest AppImage) and latest version of this repo as of Jan 28.
Using fast-csg and Manifold libraries to render, but this issue happens in both preview and in render.

Style "skeletonized" will only generate properly if you uncheck magnet holes and set hole style to "none".
Otherwise, I just get the regular "weighted" baseplate.

No errors are generated and rendering works fine, it just isn't skeletonized. 

Unfortunately I have zero skills with OpenSCAD to debug it, I only know how to use standard parametric CAD software.

jpypi, 2024-04-22:
Have you tried this since 36345f3 (current main)? I was having this issue as well, but it seems to be resolved now.

evranch, 2024-04-22:
Just tested it - working now, thanks

closed 2024-04-22 by evranch