kennetek/gridfinity-rebuilt-openscad #165: Several radii are not correct.

Bobbel93, 2024-02-08:
Please check the exact dimensions again.

https://gridfinity.xyz/specification/

[image]

[image]

plaes, 2024-02-16:
I can confirm that there is something incorrect, but I can't say what exactly:
https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/2f2567d7a9f1672aa4d6d5cfe998cdb76db81b61/standard.scad#L12-L17

Firstly, descriptions for `r_fo1`, `r_fo2` and `r_fo3` are wrong, these are actually diameters (but they are correctly divided by two in the code). Comparing these values to gridfinity standard, then `r_fo1` should be changed to `7.5`.

And another suspicious variable seems to be `r_c2`:
```
// upper base chamfer "radius"
r_c2 = 1.4;  // should be 2.15 / 2 ?
```

plaes, 2024-02-17:
@Bobbel93 Please check #167

EmperorArthur, 2024-02-26:
Depending on how PR #171 goes, I will try using the same method to handle the edges of the bin.  That is:

* Creating a polygon of the enlarged profile, right next to what you have circled.
* Sweeping it along a path.

I have a suspicion that one of the reasons for your issues is the way `hull()` works.  The math for the base is reversed, and can be hard to read.  I started cleaning it up, but realized that just sweeping a path would give a better result that could be easily verified as matching the profile.

EmperorArthur, 2024-03-11:
Looks like half of #169 was actually a duplicate of this one.
I've moved that and the relevant screenshots here.

# Screenshots of issue in Freecad

## Commit 2f2567d
[image]
[image]

## Commit 5df8267
[image]
[image]
[image]

EmperorArthur, 2024-03-11:
@plaes and @Bobbel93 please check #171

I've re-created the profile based on the spec.  Which lead me to adjust `r_base` to equal `r_fo1`.

I have **not** checked if this works on the old code, but it fixes the issue in that PR.

[image]

PR #171 (2024-04-18): Update Stacking Lip Profile
Profile is now based on https://gridfinity.xyz/specification/ ~~This does not change the final profile, merely how it is created.~~

Fixes #168 - Cura watertight issue.
Fixes #169 - FreeCad errors.

Edit: Also Fixes #165
closed 2024-04-18 by Ruudjhuu