kennetek/gridfinity-rebuilt-openscad #168: Bug: Cura Reports Model is not watertight, and may not print properly.

EmperorArthur, 2024-02-17:
Whenever I open this model in Cura I receive the following error

> Model gridfinity-rebuilt-bins 1x1x6.stl is not watertight, and may not print properly. Check X-Ray View and repair the model before printing it.

# Generation Parameters
## gridfinity-rebuilt-bins.scad
* Grid Size 1x1x6
* Single rectangular bin

* No tab
* No Scoop
* Magnet Holes Only 
* All others left at defaults

## OpenSCAD Preferences, Features
* fast-csg
* fast-csg-safer
* "predictible-output" Sic.

# Relevant System Information

* Windows 10
* Cura 5.6.0
* gridfinity-rebuilt-openscad Commit 5df82679c8916f60708ebbdf201bedac7b7b61c1 Fri Feb 16 23:01:40 2024 +0200
* OpenSCAD version 2024.02.16 (git 55be9cf86)

# STL File
[gridfinity-rebuilt-bins 1x1x6.zip](https://github.com/kennetek/gridfinity-rebuilt-openscad/files/14317301/gridfinity-rebuilt-bins.1x1x6.zip)

plaes, 2024-02-17:
Could you try with earlier commit `2f2567d7a9f1672aa4d6d5cfe998cdb76db81b61` ?

EmperorArthur, 2024-02-18:
Same result.

Interestingly, using FreeCAD to generate the model has multiple issues.  I will create another issue.

[gridfinity-rebuilt-bins 1x1x6 (Commit 2f2567d7a9f1672aa4d6d5cfe998cdb76db81b61).zip](https://github.com/kennetek/gridfinity-rebuilt-openscad/files/14322584/gridfinity-rebuilt-bins.1x1x6.Commit.2f2567d7a9f1672aa4d6d5cfe998cdb76db81b61.zip)

PR #171 (2024-04-18): Update Stacking Lip Profile
Profile is now based on https://gridfinity.xyz/specification/ ~~This does not change the final profile, merely how it is created.~~

Fixes #168 - Cura watertight issue.
Fixes #169 - FreeCad errors.

Edit: Also Fixes #165
commit 015daff2 (2024-04-18): Update Stacking Lip Profile

Profile is now based on https://gridfinity.xyz/specification/
This does not change the final profile, merely how it is created.

Fixes #168 - Cura watertight issue.
Paritally fixes #169 - FreeCad errors.
closed 2024-04-18 by Ruudjhuu