kennetek/gridfinity-rebuilt-openscad #177: Solid Bin with Lip Style 1/2

Zerogeist, 2024-03-25:
I'm attempting to generate solid bins (to modify in F360 for one-off items.)

When X/Y linear and cylindrical compartments are set to 0 I expect a model like the one labeled 1 - respecting the lip style variable I've selected.  This screenshot was generated with a cylindrical division off-screen, instead.

I do receive the expected output with style_lip set to 0 (see model labeled 2)
However, I received a solid bin with no lip whatsoever with style_lip set to 1 or 2 (see model labeled 3)

[image]

Are my expectations incorrect or is this not working as intended?

EmperorArthur, 2024-03-25:
That's expected, since style_lip 1 and 2 completely remove the lip.  Anything other than 0 is outside the official spec, so there's no way to know exactly how tall the edge you are asking for in 1 should be.

I can think of a few ways to potentially expose the variable you would want, but am already waiting on two other pull requests to be reviewed.

Zerogeist, 2024-03-26:
I guess I'm confused as to why a superficial lip would be generated when there are cylindrical compartments but not when those compartments are removed - is the lip in picture 1 part of the cylindrical compartment generation?

EmperorArthur, 2024-03-27:
Ok, this makes more sense.  Re-phrasing your issue:

# Bug: Cylindrical compartments create an unexpected lip when the lip should be disabled

This does appear to be a bug, and I will look into it.

EmperorArthur, 2024-03-30:
@Zerogeist did that last comment about `c_depth` solve your problem?  If so you could close the issue.  If not, what feature do you want?

Zerogeist, 2024-03-31:
I would really appreciate it if that variable (the superficial lip from c_depth) either made its way to style_lip or was otherwise usable with compartments set to 0.

EmperorArthur, 2024-04-02:
> I would really appreciate it if that variable (the superficial lip from c_depth) either made its way to style_lip or was otherwise usable with compartments set to 0.

I'll see what I can do, but am currently waiting on two pull requests to be reviewed.

jpypi, 2024-04-24:
I don't think it makes sense for `c_depth` to relate to `style_lip`, but I do think it might make sense to change the default to `0`. This confused me as well when I first started using this project.