kennetek/gridfinity-rebuilt-openscad #178: Printable labels

SIGSTACKFAULT, 2024-04-02:
as in https://www.printables.com/model/592545-gridfinity-bin-with-printable-label-by-pred-parame

CullenJWebb, 2024-06-03:
I second this request, and humbly request that the click labels I designed also be considered for integration: https://makerworld.com/en/models/446624#profileId-353132

goldfish1974, 2025-04-15:
I'm about to fork this repo and make this change myself. I'll do a pull request once I'm done back to this repo.

I plan on implementing a recess on the label tab that will fit the labels as mentioned by @SIGSTACKFAULT .

@CullenJWebb - The Gridfinity Generator has the Cullenect Labels which is on the Gridfinity Extended Tab too. Can you please explain what the difference is between Label Width of 1u vs 2u on the Cullenect Label tab Vs specifying a click label on a Gridfinity Extended bin that is 2u wide? The click mechanism is different. @CullenJWebb - Are you OK if I grab the code from your repo as part of this forthcoming pull request?

CullenJWebb, 2025-04-18:
@goldfish1974 You can absolutely use the code in my repo, the license allows you to do whatever you want without my permission, but I suggest using the Makerworld version (same license) as I recently added a lot of updates to the icons / available labels and haven't updated the repo yet. The label size/shape are the same.

As for the 1U vs 2U click mechanism being different it's because backwards compatibility with Cullenect V1 is only possible for 1U wide labels. In version 1 the click mechanism was 3 ribs centered on the label, and 1U width was the only supported width. With version 2 of Cullenect labels any width of label is possible because the ribs were added to the sides of the socket. I didn't want to break compatibility with bins people had already printed so 1U labels are (optionally) generated with the same 3 slots for the ribs in addition to the new side-mounted mechanism.

If by "generator" you are referring to the [perplexing labs bin generator](https://gridfinity.perplexinglabs.com/pr/gridfinity-extended/0/0) please note they he hasn't updated the version of Gridfinity Extended in a while. The latest version of Gridfinity Extended allows for bins to be generated with the V2 Cullenect socket of any width, but you'll need to download it and run it locally or use the version on MakerWorld.

goldfish1974, 2025-04-24:
@CullenJWebb - Yes, I was referring to the perplexing labs bin generator.

I'll look at implementing V2 labels as that makes most sense in regard to gridfinity-rebuilt bins. If you are making a new bin, then it is also likely that a new label would be printed also. In that case, print any new labels as V2 labels.

If a wide label is required under V2, then that wouldn't really be possible (or compatible with V1 Cullenect labels). You're basically stuck with 42mm wide compatible labels under V1.

I'll implement this as a separate pull request. First up is fixing the issue with the Label being lowered by the previous code and the miss match with radii for the block cutter.

In my experience, mixing and matching different radii and corners is difficult to do. I'm going take a look at swapping the fillet_cutter() module with a module that does the minkowski() for preserve all the curves around the corners and bottom edges of the cutout, but replace the top edge with a fillet. Using a fillet will allow for a nicer fit on the underside of the stacking lip, whilst also fitting into the corners better (N.B. I may need to adjust the vertical corner radii also, I'll see how it looks first cut)