kennetek/gridfinity-rebuilt-openscad #180: gridfinity-spiral-vase.scad: Assertion 'is_num(height_mm)' failed

vaz-ar, 2024-04-22:
I tried opening `gridfinity-spiral-vase.scad` in openSCAD and I got the following error:

~~~
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
[ERROR: Assertion 'is_num(height_mm)' failed in file gridfinity-rebuilt-utility.scad, line 419](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/419,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'assert' in file gridfinity-rebuilt-utility.scad, line 419](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/419,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad)
[TRACE: call of 'profile_wall(height_mm = undef)' in file gridfinity-rebuilt-utility.scad, line 418](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/418,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'profile_wall' in file gridfinity-spiral-vase.scad, line 364](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/364,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-spiral-vase.scad)
[TRACE: called by 'if' in file gridfinity-spiral-vase.scad, line 364](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/364,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-spiral-vase.scad)
[TRACE: called by 'children' in file gridfinity-rebuilt-utility.scad, line 439](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/439,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'children' in file generic-helpers.scad, line 161](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/161,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/generic-helpers.scad)
[TRACE: called by 'linear_extrude' in file generic-helpers.scad, line 160](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/160,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/generic-helpers.scad)
[TRACE: called by 'multmatrix' in file generic-helpers.scad, line 159](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/159,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/generic-helpers.scad)
[TRACE: called by 'for' in file generic-helpers.scad, line 158](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/158,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/generic-helpers.scad)
[TRACE: called by 'union' in file generic-helpers.scad, line 156](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/156,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/generic-helpers.scad)
[TRACE: call of 'sweep_rounded(width = 33.999, length = 33.999)' in file generic-helpers.scad, line 121](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/121,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/generic-helpers.scad)
[TRACE: called by 'sweep_rounded' in file gridfinity-rebuilt-utility.scad, line 438](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/438,C:/Users/Arnaud/Documents/_Workspace/gridfinity-rebuilt-openscad/gridfinity-rebuilt-utility.scad)
~~~

looking at the code I saw that `profile_wall` is called without any parameter (line 364)

~~~scad
                if (enable_lip) profile_wall();
                else profile_wall2();
~~~

and it looks like there are other issues in the file, like a call to `profile_wall_sub` that is not a defined module.

jpypi, 2024-04-24:
Looks like @EmperorArthur's PR https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/175 changed the definition of `profile_wall` to take a parameter and removed `profile_wall_sub`...

EmperorArthur, 2024-04-26:
> Looks like @EmperorArthur's PR https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/175 changed the definition of `profile_wall` to take a parameter and removed `profile_wall_sub`...

Downside of not using an IDE.  Global search/replace is only per file, and will do partial words, so I avoid it.

It may be worth it to see about creating a github action to exercise some more of the code.

One of my goals is to slowly remove the need for non constant global variables, which is why that was added. At the least assertions help to find the problem sooner.

EmperorArthur, 2024-04-27:
I am re-working PR #179 with some simple test cases .  Which can then be expanded in the future to make sure I don't break anything else.

I am almost done and will work on fixing this issue next.

EmperorArthur, 2024-08-06:
PR #181 fixes this, but current settings require a review before it can be merged.

Splarkszter, 2024-08-08:
> PR #181 fixes this

Thanks.
I downloaded the modified file manually and saved it to the project folder, it kinda works now.

PR #181 (2024-08-09): Fix spiral vase mode
* Fixed assertion relating to walls.
* Switched to profile_wall instead of profile_wall_sub.
* Adjusted block_pinch offset so the stacking lip is properly printed.

Fixes #180

Left is new, right is old.
[image]
closed 2024-08-09 by EmperorArthur