kennetek/gridfinity-rebuilt-openscad #192: Exported bin height dimensions inaccurate

GCheung55, 2024-06-03:
A bin for 1x1x2 Bin, with Zach's method, exports incorrect dimensions. I exported in 3MF and STL. The latest commit I'm pulling from is 36345f3.

Expected:  ~41.5mm x ~41.5mm x ~17.8mm
Actual: ~41.5mm x ~41.5mm x ~17.5

Settings:
```
/* [General Settings] */
gridx = 1;
gridy = 1;
gridz = 2;

/* [Linear Compartments] */
divx = 1;
divy = 1;

/* [Cylindrical Compartments] */
cdivx = 0;
cdivy = 0;
// orientation
c_orientation = 2; // [0: x direction, 1: y direction, 2: z direction]
cd = 10;
ch = 1;
c_depth = 1;
c_chamfer = 0.5;

/* [Height] */
gridz_define = 0; // [0:gridz is the height of bins in units of 7mm increments - Zack's method,1:gridz is the internal height in millimeters, 2:gridz is the overall external height of the bin in millimeters]
height_internal = 0;
enable_zsnap = false;

/* [Features] */
style_tab = 1; //[0:Full,1:Auto,2:Left,3:Center,4:Right,5:None]
style_lip = 0; //[0: Regular lip, 1:remove lip subtractively, 2: remove lip and retain height]
scoop = 1; //[0:0.1:1]
only_corners = false;

/* [Base] */
style_hole = 4; // [0:no holes, 1:magnet holes only, 2: magnet and screw holes - no printable slit, 3: magnet and screw holes - printable slit, 4: Gridfinity Refined hole - no glue needed]
div_base_x = 0;
div_base_y = 0;
```

EmperorArthur, 2024-06-15:
Almost certainly caused by the style_lip calculation.  Try adjusting that field and looking at the results.

I have  some work in a personal branch to measure the resulting height, but am waiting on my current PRs to be reviewed/merged.

GCheung55, 2024-06-28:
@EmperorArthur if I understand you correctly, I should try adjusting the `h_lip` to change the height of the lip.

When I change the `h_lip` to 4.4, the height of the bin + lip becomes 17.8mm. So this works, and seems like a bug. @kennetek mind chiming in here?

On a separate note, according to the [specs](https://gridfinity.xyz/specification), the bin 1x1x2 bin + lip is supposed 18.4. So what's actually correct, 17.8mm or 18.4mm?

EmperorArthur, 2024-08-04:
@GCheung55 What you're missing is that the spec expects a sharp edge at the lip.  We fillet/radius the edge.  Which, slightly reduces the bin height, but should **not** impact stacking.

The values you see are hard-coded to produce the rounded edge. I have some experimental work to auto-calculate that rounding compared to the current hard-coded value.

GCheung55, 2024-08-04:
For reference, we discussed some of this in https://github.com/kennetek/gridfinity-rebuilt-openscad/discussions/197.

@EmperorArthur Regarding to what you said about fillet/radius, then the [code comment](https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/3f2c6e28c510236571083d43890d9ca35632b58b/gridfinity-rebuilt-bins.scad#L16) should be corrected to be 17.5515mm and not 17.8, right?

closed 2025-06-26 by EmperorArthur