kennetek/gridfinity-rebuilt-openscad #193: Can bins `gridz` param support decimals?

GCheung55, 2024-06-03:
Bins `gridz` parameter currently does not support decimals. There is a computation error where the bins height does not add up to an expected height, for example 1x1x2 bin should have a height of 17.8mm, however, the actual result is 17.5mm. 

Making the `gridz` support decimal places, corresponding to setting the `gridz define` to be overall external height, can help compensate.

In my testing, changing `gridz` parameter at line 36 of `gridfinity-rebuilt-bins.scad` to the following supports decimals. Note that it could introduce rounding issues if `gridz define` is not anything besides overall external height.
```
gridz = 6; // .01
```

EmperorArthur, 2024-09-01:
Please note that the resulting height difference is because we deliberately round the bin's lip instead of bringing it to a sharp point.

I cannot emphasize this enough.  **THE BIN HEIGHT DIFFERENCE DOES NOT IMPACT COMPAITBILITY WITH OTHER GRIDFINITY BINS!**  It is an artifact of how we implement the spec, and can safely be ignored.

GCheung55, 2024-09-02:
Hey @EmperorArthur. Thanks for responding. I filed this a while ago and we had already discussed the issue about heights and stacking in other tickets and discussion thread. Indeed, the rounded lip does not interfere with stacking nor does it affect the stacking height. I’ll close this ticket.

closed 2024-09-02 by GCheung55

EmperorArthur, 2024-09-02:
I have created a pull request that will hopefully prevent others from experiencing the same confusion.  I was never a fan of the magic constant, even when I knew how it was calculated.